esp8266/Arduino #2830: HTTPClient::PATCH missing

Heizelmann, 2017-01-04:
Because HTTPClient::PATCH is missing in the library
I tried it with the basic function sendRequest(), but then I get an connection refused error. 
```
HTTPClient httpClient;
httpClient.begin(url);
httpClient.setAuthorization(PIMATIC_USER, PIMATIC_PWD);
int httpCode = httpClient.sendRequest("PATCH", content);
  if (httpCode > 0) {
    Serial.printf("[HTTP] PATCH... code: %d\n", httpCode);
  } else {
    Serial.printf("[HTTP] GET... failed, error: %s\n", httpClient.errorToString(httpCode).c_str());
  }
  httpClient.end();
```

For comparison a low level solution with WiFiClient works on same server.
``` 
WiFiClient client;
...
client.print("PATCH " + url + " HTTP/1.1\r\n" +
               "Authorization: Basic " + unameenc + "\r\n" +
               "Host: " + HOST_IP + "\r\n" +
               "Content-Type: application/json" + "\r\n" +
               "Content-Length: " + content.length() + "\r\n\r\n" +
               content);
```

devyte, 2017-10-04:
@Heizelmann please try latest git, which has PR #3590  merged.
Please report your results back here.

Heizelmann, 2017-10-11:
Thanks a lot! At a glance it looks good. Currently I have no time to test it. I will do it as soon as possible.

Heizelmann, 2017-10-11:
As far as I can see it is only a wrapper around the send request function I already tested. I did a short retry with the latest version from github but the error is still the same.

igrr, 2017-10-11:
Can you please attach the complete sketch which reproduces the issue?

Heizelmann, 2017-10-12:
While preparing the example code for posting I detected my fault. In the example above the variable url can not be the same for WifiClient and HttpClient. For WifiClient it is only the URL extension whithout "http://myIP". With HttpClient it must be the full url.

` httpClient.begin("http://" + String(HOST_IP) + `url);`

Everything works fine with the new PATCH function now. The issue can be closed. Thanks a lot and excuse my fault. Would be nice if you can inform me when the new feature will be released.

devyte, 2017-10-12:
Closing as resolved.

closed 2017-10-12 by devyte