esp8266/Arduino #2831: SoftwareSerial::Write() does not use inverse logic

titos-carrasco, 2017-01-05:
### Basic Infos

#### Hardware
Hardware: NodeMcu

### Description

In file SoftwareSerial.cpp, SoftwareSerial::write(uint8_t b) does not use "inverse logic".

I changed some lines...

### Sketch

```cpp
size_t SoftwareSerial::write(uint8_t b) {
   if (!m_txValid) return 0;

   if (m_invert) b = ~b;
   // Disable interrupts in order to get a clean transmit
   cli();
   if (m_txEnableValid) digitalWrite(m_txEnablePin, HIGH);
   unsigned long wait = m_bitTime;
   digitalWrite(m_txPin,!m_invert ? HIGH : LOW);
   unsigned long start = ESP.getCycleCount();
    // Start bit;
   digitalWrite(m_txPin, !m_invert ? LOW : HIGH );
   WAIT;
   for (int i = 0; i < 8; i++) {
     digitalWrite(m_txPin, (b & 1) ? HIGH : LOW);
     WAIT;
     b >>= 1;
   }
   // Stop bit
   digitalWrite(m_txPin, !m_invert ? HIGH : LOW);
   WAIT;
   if (m_txEnableValid) digitalWrite(m_txEnablePin, LOW);
   sei();
   return 1;
}
```

devyte, 2017-10-04:
@titos-carrasco per the lib properties, the repo for the ESP's SoftwareSerial is https://github.com/plerup/espsoftwareserial
You may want to open an issue/PR over there, then we can update the lib here.

@igrr I am using latest git, and the SoftwareSerial lib is not in the directory structure. However, it is present under the 2.3.0 release directory structure. I assume that means that it gets packaged in the release process. Exactly how is it pulled in, and which version is used?

devyte, 2018-07-13:
This issue is for SoftwareSerial. Similar issue for hw serial: #4896 .

devyte, 2019-10-17:
cc @dok-net

devyte, 2019-11-10:
No feedback in a long time.
I don't know if relevant, but I see an inverse flag available in the begin() signature.
Closing.

closed 2019-11-10 by devyte