esp8266/Arduino #2833: class ArduinoOTAClass' has no member named 'getCommand'

tiestvangool, 2017-01-06:
Team,

I am unable to compile the firmware for the ESP8266 using Platformio. After downloading the ArduinoOTA, I consistently receive the error log below, can someone tell me what I am doing wrong?

Thanks in advance,
-Tiest

----- ERROR LOG -----

tiest@chrx:~/pio/ota$ curl -o src/main.ino https://raw.githubusercontent.com/esp8266/Arduino/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1945  100  1945    0     0   3672      0 --:--:-- --:--:-- --:--:--  3676
tiest@chrx:~/pio/ota$ pio run --target upload
[Thu Jan  5 21:29:57 2017] Processing nodemcuv2 (platform: espressif8266, board: nodemcuv2, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting main.ino
Collected 23 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ArduinoOTA> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
Compiling .pioenvs/nodemcuv2/src/ArduinoOTA.o
Compiling .pioenvs/nodemcuv2/src/main.ino.o
Archiving .pioenvs/nodemcuv2/libFrameworkArduinoVariant.a
Indexing .pioenvs/nodemcuv2/libFrameworkArduinoVariant.a
Compiling .pioenvs/nodemcuv2/FrameworkArduino/Esp.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/FS.o
/home/tiest/pio/ota/src/main.ino: In lambda function:
/home/tiest/pio/ota/src/main.ino:35:20: error: 'class ArduinoOTAClass' has no member named 'getCommand'
if (ArduinoOTA.getCommand() == U_FLASH)
^
src/ArduinoOTA.cpp: In constructor 'ArduinoOTAClass::ArduinoOTAClass()':
src/ArduinoOTA.cpp:34:3: error: class 'ArduinoOTAClass' does not have any field named '_rebootOnSuccess'
, _rebootOnSuccess(true)
^
src/ArduinoOTA.cpp: At global scope:
src/ArduinoOTA.cpp:95:60: error: no 'void ArduinoOTAClass::setPasswordHash(const char*)' member function declared in class 'ArduinoOTAClass'
void ArduinoOTAClass::setPasswordHash(const char * password) {
^
src/ArduinoOTA.cpp:101:53: error: no 'void ArduinoOTAClass::setRebootOnSuccess(bool)' member function declared in class 'ArduinoOTAClass'
void ArduinoOTAClass::setRebootOnSuccess(bool reboot){
^
*** [.pioenvs/nodemcuv2/src/main.ino.o] Error 1
src/ArduinoOTA.cpp: In member function 'void ArduinoOTAClass::_runUpdate()':
src/ArduinoOTA.cpp:317:8: error: '_rebootOnSuccess' was not declared in this scope
if(_rebootOnSuccess){
^
src/ArduinoOTA.cpp: At global scope:
src/ArduinoOTA.cpp:345:33: error: no 'int ArduinoOTAClass::getCommand()' member function declared in class 'ArduinoOTAClass'
int ArduinoOTAClass::getCommand() {
^
*** [.pioenvs/nodemcuv2/src/ArduinoOTA.o] Error 1

arenddeboer, 2017-04-29:
This is because PlatformIO ships with outdated (Arduino) libraries.
I just ran into this again after setting up a new laptop.
What I did was adding a lib_extra_dirs to the PlatformIO platformio.ini project file like:

```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =
  SPI
  OneWire
  DallasTemperature
lib_extra_dirs = /mnt/data/PlatformIOProjects/MeshPirClient/lib_esp8266
```

Inside this lib_esp8266 dir I have a folder called ArduinoOTA copied from the latest Arduino library (https://github.com/esp8266/Arduino/tree/master/libraries/ArduinoOTA)

airdrummingfool, 2017-05-20:
This is discussed [on the Platform.io Community](https://community.platformio.org/t/class-arduinootaclass-has-no-member-named-getcommand/1204/6). Quote:
> That is not our problem. Need to wait when new version of Arduino Core for ESP8266 will be released. Nevertheless, you can use temporary "development" version:
> 
> http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version70

#3020 is a duplicate of this issue.

earlephilhower, 2019-01-16:
#2445 added getCommand many years ago.

closed 2019-01-16 by earlephilhower