esp8266/Arduino #2834: HTTPS POST request to api returns -1 response  (ssl)

thorburn1, 2017-01-06:
I am trying to post a .json message with field "chip_id" and "power" to my https api made. But response error code -1 is coming back, and data is not getting inserted there.  
I used this code found and modified the url , wifi credentials ,and thumbprint.
Can someone explain the issue?
(working fine with postman)  
(using standalone esp8266)
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k(64k SPIFFS)
CPU Frequency:		80Mhz
DEBUG   LEVEL            all                 
debug port                  disabled
Flash Frequency:	40Mhz
Upload Using:		?OTA / SERIAL?
Reset Method:		ck 
arduino as isp

### Sketch
#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>

void setup() {

Serial.begin(115200);                                 
WiFi.begin("iot", "oksurewhynot");   

while (WiFi.status() != WL_CONNECTED) {  //Wait for the WiFI connection completion

delay(500);
Serial.println("Waiting for connection");

}

}

void loop() {

if(WiFi.status()== WL_CONNECTED){   //Check WiFi connection status

HTTPClient http;
  http.begin("www.fusmin.com/hooterstatus/api/HooterStatusUpdate.php" ,  "‎F6 55 EE 84 54 DA CF 23 82 90 F4 BA D2 F5 30 AA FC B8 D3 7A");
  http.addHeader("Content-Type", "application/json");
  String postMessage = String("{'chip_id' : 'rohitesp', 'power' : '2'}");
  int httpCode = http.POST(postMessage);
  Serial.print("http result:");
  Serial.println(httpCode);
  http.writeToStream(&Serial);
  String payload = http.getString();
  http.end();

}else{

Serial.print("Error in Wifi connection");   

}

delay(30000);  //Send a request every 30 seconds

}

### Debug Messages

Waiting for connection
[HTTP-Client][begin] url: www.fusmin.com/hooterstatus/api/HooterStatusUpdate.php
[HTTP-Client][begin] failed to parse protocol
[HTTP-Client] connect: HTTPClient::begin was not called or returned error
[HTTP-Client][returnError] error(-1): connection refused
http result:-1
[HTTP-Client][returnError] error(-4): not connected
[HTTP-Client][returnError] error(-4): not connected
[HTTP-Client][end] tcp is closed
pm open,type:2 0

vicnevicne, 2017-01-06:
Well, the first error your log says is "**[begin] failed to parse protocol**".
And indeed, the URL you pass to begin does not start with a protocol ("http://"), so begin fails.
After that, connect complains that begin returned an error and so on, so I would suggest you to first fix the url by adding "http://" and retry...

thorburn1, 2017-01-06:
ok , so I edited my uri , but still http response is -1 , debug message i updated in the original post.
My thumbprint is correct, cross checked it from the certificate(sha1)

vicnevicne, 2017-01-06:
Wow, modifying your original code and log along the way is not very good practice because the discussion is now impossible to follow for others.
Anyway, I have zero experience with SSL on ESP, so I'm afraid it's now out of my field.
However, do you have success if using a GET method ? Just to know if it is really POST-related or if you have to look somewhere else...

thorburn1, 2017-01-06:
I think its something to do with my certificate instead ? code works fine with other api's out there , or with GET i guess, even am very new to this.

thorburn1, 2017-01-06:
debug message : 
Waiting for connection
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:wr
:sent 73
:rn 1370
:ww
:rd 5, 1370, 0
:rdi 1370, 5
:rd 74, 1370, 5
:rdi 1365, 74
State:	receiving Server Hello (2)
:rd 5, 1370, 79
:rdi 1291, 5
:rd 1286, 1370, 84
:rdi 12:rch 1370, 1370
:rch 2740, 1370
:rch 4110, 749
86, 1286
:c 1286, 1370, 4859
:rd 3480, 3489, 0
:rdi 1370, 1370
:c 1370, 1370, 3489
:rdi 1370, 1370
:c 1370, 1370, 2119
:rdi 749, 740
State:	receiving Certificate (11)
:rd 5, 749, 740
:rdi 9, 5
:rd 4, 749, 745
:rdi 4, 4
:c0 4, 749
State:	receiving Server Hello Done (14)
State:	sending Client Key Exchange (16)
:wr
:sent 267
:ww
:wr
:sent 6
:ww
State:	sending Finished (16)
:wr
:sent 69
:rn 75
:ww
:rd 5, 75, 0
:rdi 75, 5
:rd 1, 75, 5
:rdi 70, 1
:rd 5, 75, 6
:rdi 69, 5
:rd 64, 75, 11
:rdi 64, 64
:c0 64, 75
State:	receiving Finished (16)
:ref 2
pos:0 len:65 invalid hex sequence: â€
:ur 2
:wr
:sent 53
:rn 53
:rcl
:abort
:ww
Alert: close notify
:ur 1
:del
http result:-1
pm open,type:2 0

pieman64, 2017-01-06:
@thorburn1 you are lucky that I have been doing some fingerprint stuff in the last few days. After a **very minor** mod to your sketch Serial Monitor shows:

```
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Fri, 06 Jan 2017 10:52:26 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Apache/2.2.31 (Amazon)'
[HTTP-Client][handleHeaderResponse] RX: 'X-Powered-By: PHP/5.5.30'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 106'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=UTF-8'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 106
http result:200
```
OK so what's the mod. For some reason you have an invisible character at the start of your fingerprint and when I pulled down your fingerprint from https://www.grc.com/fingerprints.htm it seems to have given me the same. Simply go to the position in your sketch before the first F in the fingerprint and watch as you do a backspace on the keyboard. It will remove the invisible character and you will be good to go.

thorburn1, 2017-01-06:
@pieman64  That solved the problem I was fighting for since like 9 hours  , thanks a lot.Never expected such error 💃

closed 2017-01-06 by thorburn1

thorburn1, 2017-01-06:
//INCASE SOMEONE IN FUTURE WANTS, THIS IS THE WORKING CODE TO SEND JSON DATA TO HTTPS API USING ESP8266 (SSL)

#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>

void setup() {

Serial.begin(115200);                                 
WiFi.begin("iot", "oksurewhynot");   

while (WiFi.status() != WL_CONNECTED) 

delay(500);
Serial.println("Waiting for connection");

}

void loop() {

if(WiFi.status()== WL_CONNECTED){   //Check WiFi connection status

HTTPClient http;
  http.begin("https://fusmin.com/hooterstatus/api/HooterStatusUpdate.php" ,  "F6 55 EE 84 54 DA CF 23 82 90 F4 BA D2 F5 30 AA FC B8 D3 7A‎");
  http.addHeader("Content-Type", "application/json");
  String postMessage = "{\"chip_id\":\"rohitesp\", \"power\":\"2\"}";
  int httpCode = http.POST(postMessage);
  Serial.print("http result:");
  Serial.println(httpCode);
  http.writeToStream(&Serial);
  String payload = http.getString();
  http.end();

}else{

Serial.print("Error in Wifi connection");   

}

delay(30000);  //Send a request every 30 seconds

}

sergeykobzarenko, 2017-12-06:
it doesnt help if you just changed http to https

dan62, 2019-10-16:
> //INCASE SOMEONE IN FUTURE WANTS, THIS IS THE WORKING CODE TO SEND JSON DATA TO HTTPS API USING ESP8266 (SSL)
> 
> #include <ESP8266HTTPClient.h>
> #include <ESP8266WiFi.h>
> 
> void setup() {
> 
> Serial.begin(115200);
> WiFi.begin("iot", "oksurewhynot");
> 
> while (WiFi.status() != WL_CONNECTED)
> 
> delay(500);
> Serial.println("Waiting for connection");
> 
> }
> 
> void loop() {
> 
> if(WiFi.status()== WL_CONNECTED){ //Check WiFi connection status
> 
> HTTPClient http;
> http.begin("https://fusmin.com/hooterstatus/api/HooterStatusUpdate.php" , "F6 XX XX XX XX XX XX XX XX XX XX BA D2 F5 30 AA FC B8 D3 7A‎");
> http.addHeader("Content-Type", "application/json");
> String postMessage = "{"chip_id":"rohitesp", "power":"2"}";
> int httpCode = http.POST(postMessage);
> Serial.print("http result:");
> Serial.println(httpCode);
> http.writeToStream(&Serial);
> String payload = http.getString();
> http.end();
> 
> }else{
> 
> Serial.print("Error in Wifi connection");
> 
> }
> 
> delay(30000); //Send a request every 30 seconds
> 
> }

Worked like a charm, thanks alot. 💯

aminbaig, 2021-03-09:
Hi all, I know this issue has been closed for some time but I tried this today and I am getting -1 in http response!!!

Here is my code (practically copied and pasted):

`
#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>

void setup() {

Serial.begin(115200);
WiFi.begin("codenai2", "drx34cb1234");

while (WiFi.status() != WL_CONNECTED)

delay(500);
Serial.println("Waiting for connection");

}

void loop() {

if(WiFi.status()== WL_CONNECTED){ //Check WiFi connection status

HTTPClient http;
http.begin("https://www.livesessions.host/api/v1/readings" , "CC 27 07 64 5D 02 22 70 01 61 A1 74 67 1C B8 B0 25 8E AA 91‎");
http.addHeader("Content-Type", "application/json");
String postMessage = String("{'deviceMAC':'ddd eee fff', 'humidity':'99', 'soilTemp':'88'}");
int httpCode = http.POST(postMessage);
Serial.print("http result:");
Serial.println(httpCode);
http.writeToStream(&Serial);
String payload = http.getString();
http.end();

}else{

Serial.print("Error in Wifi connection");

}

delay(30000); //Send a request every 30 seconds

}`

would appreciate any help as I can't figure out why this solution isn't working for me.

d-a-v, 2021-03-09:
@aminbaig

The API you're using is deprecated.

Please
- update to the latest version
- check the examples namely HelloServerBearSSL (File>examples>esp8266httpclient>)
- open a new issue if there's a problem with the proposed examples

aminbaig, 2021-03-18:
Hi @d-a-v , thank you for your reply and apologies for the delayed response.

I have updated as advised by you and gone through the examples. I am confused about which of the examples to use as my scenario is as follows:
1. I have an https server in nodejs and mongodb at the backend.
2. I am trying to send  a post request from the dht11 readings from wemos d1 mini to this https web server

I can't seem to find an example with the BearSSL to do that.

Hope you can help.