esp8266/Arduino #2836: NodeMCU 12E v3 Resets after 30-45 Minutes.

jaajee, 2017-01-06:
I have little different issue with Lolin NodeMCU 12-E V3. If I upload Blink or Push Button Example and run it all day long I do not get any error. But if I upload the following sketch (from https://github.com/torinnguyen/ESP8266Wemo/blob/master/ESP8266WeMo.ino). It starts without any issues but after 30-45 minutes it resets/ or reboots and gives me (ets Jan 8 2013,rst cause:2, boot mode:(3,6) wdt reset) message and starts running again (I don't do anything to make it run again). Again after 30-45 minutes same thing Resets/or Reboots and starts again. I tried many different solutions posted on this website but no luck. Someone suggested on http://community.blynk.cc/t/solved-esp8266-nodemcu-v1-0-and-wdt-resets/7047/11
 that adding: 
**ESP.wdtDisable();** **ESP.wdtEnable(WDTO_8S);** in Setup(); 
and 
**ESP.wdtFeed();** in Loop(); should fix it but again no luck.
I don't think its a power supply issue because it runs smoothly for 30-45 minutes.
Any Help will be appreciated. Thanks in Advance.

----------------------------- ---------------------------------------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E v3
Core Version:      	1.6.5 & 1.6.7

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck / nodemcu

### Sketch

```#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUDP.h>

const char* ssid = "---";                       // your network SSID (name)
const char* pass = "---";                       // your network password

const char* friendlyName = "Arduino";           // Alexa will use this name to identify your device
const char* serialNumber = "221517K0101768";                  // anything will do
const char* uuid = "904bfa3c-1de2-11v2-8728-fd8eebaf492d";    // anything will do

// Multicast declarations
IPAddress ipMulti(239, 255, 255, 250);
const unsigned int portMulti = 1900;
const unsigned int webserverPort = 9876;

// Witty Cloud Board specifc pins
const int LED_PIN = 13;

//-----------------------------------------------------------------------
//-----------------------------------------------------------------------

//int status = WL_IDLE_STATUS;

WiFiUDP Udp;
byte packetBuffer[512]; //buffer to hold incoming and outgoing packets

ESP8266WebServer server(webserverPort);

//-----------------------------------------------------------------------
// UDP Multicast Server
//-----------------------------------------------------------------------

char* getDateString()
{
  //Doesn't matter which date & time, will work
  //Optional: replace with NTP Client implementation
  return "Wed, 29 Jun 2016 00:13:46 GMT";
[... 209 lines cut ...]
  Serial.print(ipMulti);
  Serial.print(":");
  Serial.println(portMulti);
  
  //Web Server
  server.on("/", handleRoot);
  server.on("/setup.xml", handleSetupXml);
  server.on("/upnp/control/basicevent1", handleUpnpControl);
  server.onNotFound(handleNotFound);
  server.begin();
  Serial.print("HTTP server started on port ");
  Serial.println(webserverPort);
}

void loop()
{
  UdpMulticastServerLoop();   //UDP multicast receiver
  server.handleClient();      //Webserver
  ESP.wdtFeed();
}

```

```
```

witnessmenow, 2017-01-08:
I had a quick look through the code and I didn't see anything obvious (to me)

I have a [library to do this](https://github.com/witnessmenow/esp8266-alexa-wemo-emulator) (emulate a wemo for alexa), I havent had any issues with it restarting. Maybe give it a go, check out the examples to see how to use it.

jaajee, 2017-01-09:
witnessmenow, thanks for yourreply. I will check out and give it a try and let you know. Again thanks

jaajee, 2017-01-11:
@witnessmenow thanks man , it worked and its been running for last two days without a single reboot or resets. 
I have question though. Do you think somehow I could make it work with WEMOs app? I know its an emulator but have you tried to tweak it to make it work with WEMO app.

witnessmenow, 2017-01-12:
Hey, no I didn't really. I just downloaded the app and it didn't work. Maybe it's possible, but no idea what's involved

jaajee, 2017-01-12:
Since my initial issue has be resolved I am going to close this issue.

closed 2017-01-12 by jaajee