esp8266/Arduino #2837: Webserver and WiFiClientSecure (SMTP SSL) fatal error 29

Marius58, 2017-01-06:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

SORRY THE DECODER DOES NOT WORK 

Exception (29):
epc1=0x4023eb0b epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000c depc=0x00000000

ctx: cont 
sp: 3fff23f0 end: 3fff2c20 offset: 01a0

>>>stack>>>
3fff2590:  3fffb4c0 00000044 3fff29a4 4023eb08  
3fff25a0:  00000100 00000100 3fffb6a4 4023f043  
3fff25b0:  3fffb4c0 00000100 3fffb18c 40240943  
3fff25c0:  3fffb6a4 3fff4444 00000003 402445df  
3fff25d0:  3fff1ba8 00000200 00000200 00000000  
3fff25e0:  00000100 3fff984e 3fff29a4 40244aec  
3fff25f0:  3fff4444 3fffb6a4 3fff29a4 4024497c  
3fff2600:  3fffb494 00000026 3fff29a4 4024481c  
3fff2610:  00000000 0000003c 3fff984e 00000004  
3fff2620:  00000000 3fffb494 3fff984e 4023d78c  
3fff2630:  00000288 00000000 65b62cea d20baa30  
3fff2640:  a9a9f21a f25aff4c aaab75b3 90598882  
3fff2650:  42a9754c 8d118b03 556c751f 93f12f84  
3fff2660:  47cc899b a8bc35e4 00000080 00000000  
3fff2670:  00000000 00000000 00000000 00000000  
3fff2680:  00000000 00000000 00000000 00000000  
3fff2690:  00000000 80260000 feefeffe feefeffe  
3fff26a0:  feefeffe feefeffe feefeffe feefeffe  
3fff26b0:  feefeffe feefeffe feefeffe feefeffe  
3fff26c0:  feefeffe feefeffe feefeffe feefeffe  
3fff26d0:  feefeffe feefeffe feefeffe feefeffe  
3fff26e0:  feefeffe feefeffe feefeffe feefeffe  
3fff26f0:  feefeffe feefeffe feefeffe feefeffe  
3fff2700:  feefeffe feefeffe feefeffe feefeffe  
3fff2710:  feefeffe feefeffe feefeffe feefeffe  
3fff2720:  feefeffe feefeffe feefeffe feefeffe  
3fff2730:  feefeffe feefeffe feefeffe feefeffe  
3fff2740:  feefeffe feefeffe feefeffe feefeffe  
3fff2750:  feefeffe feefeffe feefeffe feefeffe  
3fff2760:  feefeffe feefeffe feefeffe feefeffe  
3fff2770:  feefeffe feefeffe feefeffe feefeffe  
3fff2780:  feefeffe feefeffe feefeffe feefeffe  
3fff2790:  feefeffe feefeffe feefeffe 40310000  
3fff27a0:  00000001 0000001c 3ffe82f4 401004f4  
3fff27b0:  feefeffe 00000002 3ffe82f4 402017e4  
3fff27c0:  00000046 40348000 00000000 4021698e  
3fff27d0:  00000008 00000000 00000020 40100ece  
3fff27e0:  3ffeb518 40100b79 00000002 00000100  
3fff27f0:  00000005 00000008 40100a2e 00000002  
3fff2800:  00000002 4010082f 00000002 3ffef660  
3fff2810:  00000001 00000000 00000020 40100ece  
3fff2820:  00000005 00000000 00000020 40100ece  
3fff2830:  3ffeb515 401041ab 3ffeefa0 00000000  
3fff2840:  40101be1 3ffeefa0 00000000 00000002  
3fff2850:  00007fff 044dd1d1 3ffef9c8 40101d70  
3fff2860:  3ffebd9c 00000000 00000000 00000002  
3fff2870:  00000000 044dd1d1 401021ae 00000100  
3fff2880:  7fffffff 3ffebd9c 3ffebd9c 00000001  
3fff2890:  00000005 00000000 00000020 40100ece  
3fff28a0:  3ffeb515 401041ab 3ffeef50 4000050c  
3fff28b0:  00000000 00000000 0000001f 40104cad  
3fff28c0:  4000050c 044dd8a0 3ffef9c8 40101d70  
3fff28d0:  40242cc9 00000030 00000019 ffffffff  
3fff28e0:  40243a9c 0cf2f005 dd9e0761 a89e7050  
3fff28f0:  d50cd43a b0c536e5 e0761dd9 93cb4680  
3fff2900:  cfbcac30 f737ad03 b6af3b12 9a70c509  
[... 436 lines cut ...]
// #=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#
void loop(void){
  server.handleClient();

// About one seconde delay--------------------------------------------
  if (second()!=SecLast)
    {
		SecLast=second();
		if (Mail_id != 0)
      {
      SendMail(Mail_id); // send email if required
      Mail_id=0;
      }
    }
}
    
```

### Debug Messages

```
messages here
Thank you for your help.

```

Palatis, 2017-01-07:
probably out of memory.

axTLS needs 16k ram to run, plus 1.6k buffer for MTU.
and sometimes when changing keys it needs twice as much.

Marius58, 2017-01-07:
Hello,
Thank you for your comment.
I suspected a memory problem. But the web server and SSL email each individually working.
Only in the case that the webserser responds to a request the SSL email does not work after this event.

Palatis, 2017-01-07:
yes, because when they work one at the time there isn't as much memory pressure.

Marius58, 2017-01-08:
OK,
In this case I have to use only the standard SMTP without SSL which works well.
Thanks.

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
Closing.

closed 2018-05-29 by devyte