esp8266/Arduino #2840: running Server object on Soft AP

mushfiq-cse05-buet, 2017-01-07:
hello guys, i'm running a WiFiServer object on my esp01 but this time i opened up a Soft AP on it so that i dont need any separate WiFi Router to connect to my esp01. i'm using a simple android app to connect my esp01. the process is very simple, i take my android phone in my hand search for the SSID of Soft AP of my esp01 then providing the password my android phone get connected, then i open up a simple tcp client app to send some data or strings u  may say. according to my code my esp01 is always listening to a particular port inside the loop(). the whole system runs very smoothly without any flaw. the situation is just like this, i power up the whole system  and start sending data or string from my android app after getting the data my esp01 shows a number on seven segment display, at the beginning everything works very fine  say for first 1-5 hour. then i go to bed and sleep for 8-10 hours after wake up i take my android phone in my hand and search for the SSID of Soft AP of esp01 and i dont see that SSID anymore, can anybody tell me a good solution to this problem ???

retsifp, 2017-01-09:
- Please stay within ONE issue with the same problem (#2841 is a duplicate).
- You will need to do more debugging on your own, e.g. connecting your computer via serial cable, add some output about the current state of the module to the sketch, compile with all debug options and look what is there after crash
- **As it was suggested in the issue template**, provide more information about your project (sketch, hardware, ...)

devyte, 2017-10-04:
Duplicate of #2841 . Closing due to can't reproduce.

closed 2017-10-04 by devyte