esp8266/Arduino #2842: my esp8266 died a bit inside.. :'(

nikant, 2017-01-07:
### Basic Infos
Arduino IDE 1.6.5-r5

#### Hardware
Hardware:			ESP-12E on NodeMCU devboard
Core Version:      	2.3.0

### Description
Sorry to ask this here but I'm kind in a desperate/curious situation. I was continuing writing stuff in a sketch (ink below) and had the idea to test it with CPU set at 160MHz.. It was very responsive and fast. Done with that I reverted and flashed the same sketch at 80MHz CPU as I always do.. 
The NodeMCU devboard died a bit inside it seems.. 
WiFi signal now is extremely low no matter what I tried. When scanning for available APs it used to show 10-15 in the area and now it shows only my router at a -75dBm signal **right next to it**..!
A second NodeMCU that replaced it works fine.

**Could the 160MHz change cause a hardware problem?** It is so strange.. 

Since then I tried flashing a lot of other sketches trying to see how it would respond. Everything checks out (memory sizes, chip ids etc.) but the WiFi power.. Also tried the NodeMCU flasher loading lua scripts or wiping the board (anyone knows a good method shocking it or factory resetting?) Even with lua scripts it shows only my router and that at a low signal. 

Has anyone experienced such behavior before? Is it fried somehow..?

### Settings in IDE

Module:  			NodeMCU 1.0 ESP-12E
Flash Size: 		4MB
CPU Frequency:	80Mhz - problem occurred after 160MHz

### Sketch
I was trying this sketch https://github.com/nikant/THE433 **at 160MHz** 
(I don't think that it could cause a hardware problem.. I've used it extensively in my home automation **at 80MHz** with even more modifications and didn't have any problem from many months)

asetyde, 2017-01-07:
about voltage ? is enought ?

nikant, 2017-01-07:
I believe so. Since I run a mod of the above script for my home automation I don't trust just my laptops USB voltage. I use usb plugs 5v/1-1.5A

asetyde, 2017-01-07:
i don't know if supply is good enoubth but use at 160mhz is a new world for me , i think new issues it comes on this support library .. not good idea , 99%  use at 80mhz , i build many products with esp  core nodemcu about domotics , i never see problem , try to check your code before use mhz vitamine

liquidfalcon, 2017-01-09:
I have run many modules for extended periods of time at 160MHz (Months and months) and have never encountered any real issues. There is an internal calibration used by the SDK for WiFi that may be worth resetting. I don't remember the exact location, but, if you use esptool, there's a flag to erase the entire memory. Try doing that, and then flashing your program once again.

nikant, 2017-01-10:
I honestly believe that something is fried inside.. Maybe some mix of my code and connections overheated stuff.. 

I have flashed all kinds of code the previous days (not my code) to test it's wifi reception. It's damaged.!

I even cleared flash with flash downloader writing 1MB blanks to all addresses 

[image]

I think I should make an esp8266/nodemcu place in my garden (with an IoT cross perhaps on top..) 
:/ :(

jjssoftware, 2017-01-11:
Just an observation really - you have an ESP-12E with 4MB of flash but the screenshot of the espressif flasher shows that the flash size was selected as 4Mbit. For 4MB of flash, you'd select 32Mbit

nikant, 2017-01-12:
@jjssoftware Yes.. you're right.. tried the flasher with changing all options after the correct ones.. being desperate.. the screenshot is from one of the many tries..

devyte, 2017-10-04:
@nikant Is this an issue? Although I haven't tested it myself, use of 160MHz should work correctly.
The esptool can be used to erase the entire flash from command line, I suggest trying that. Also, please try latest git after wiping. There is rf calibration step done on startup.
Please report back here if this helps.

nikant, 2017-10-08:
Thank you @devyte for the suggestions. I have put aside the defective Nodemcu for many months now. I strongly believe that something heated up and desoldered somewhere inside, but I'm no expert. I will try your suggestions. No need to keep this thread open. 
Thank you :)

closed 2017-10-08 by nikant