esp8266/Arduino #2843: My ESP8266 connects to wifi without specifiying any SSID and password!

jumpjack, 2017-01-07:
I modified BasicHttpClient to suite my needs, but I get strange results: ESP connects to "something" but I don't know to WHAT!
In a first time I left the default SSID and pwd of the source, so I expected no connection to happen... but ESP connected to my wifi network and properly downloaded a file from internet!
Then I completely deleted SSID and pwd:
**WiFiMulti.addAP("", "");**
Well... it's still working fine!
It connects to my network and downloads file from internet!

I guess the ESP is using ssid and pwd I previously specified in other experimental sketches... but I think it is not an acceptable way  to work!

How can I get a list of previously accessed APs?
How can I get the name of currently connected network?
How can I get a list of WifiMulti() available methods ans properties?

But, above all: how can I trigger an error if I specify wrong password or non-existing SSID?

retsifp, 2017-01-09:
Well, I _think_ the `WifiMulti`-Library stores the APs inside flash and once you added an AP, it'll connect to it.
This behaviour is not documented [here](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/doc/esp8266wifi/station-examples.md), but the function `void APlistClean(void)` ([here](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.h)) could be worth a try.

Please read also [here](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/doc/esp8266wifi/station-class.md), you can read your current SSID and PSK by calling 
```
Serial.printf("SSID: %s\n", WiFi.SSID().c_str());
Serial.printf("PSK: %s\n", WiFi.psk().c_str());
```

jumpjack, 2017-01-09:
By the way, in the .h file you linked, which I already had checked, I noticed just now the WifiAPlist structure, but I don't know how to access it, I only know basics of C/C++...

Thanks for the tip about current SSID/pwd

retsifp, 2017-01-09:
`APlist` a vector of `WifiAPlist_t`. So I think it'll contail all the APs, but it's private and without some knowledge of C++-Programming, it won't be that easy to access it.

liquidfalcon, 2017-01-09:
Additionally, Espressif's internal SDK will remember the last network it connected to (Only one, though - Never more) and automatically connect to it upon boot. You can access that stored network via `WiFi.SSID()` and `WiFi.psk()`. To disable this auto-connect, call `WiFi.persistent(false)`

jumpjack, 2017-01-09:
About C++ un-knowledge.... How do I enable the DEBUG messages in WifiMulti? I see some #define but I don't know how to use them.

liquidfalcon, 2017-01-09:
Almost all of the debugging information on the Arduino core is enabled via the "Debug Level" and "Debug Port" menus on the Arduino IDE, under tools. Choose a Debug Port, then set the Debug Level to "WiFi", and recompile. 

I do believe you need to set your module to "Core Development Module" to enable the menu, however.

jumpjack, 2017-01-09:
Do I have to set the debug port to same number of serial monitor?

liquidfalcon, 2017-01-09:
Most likely, yes - It just spits out the debug data to whichever port you so choose.

devyte, 2017-10-04:
@jumpjack the SDK stores some info, including the wifi credentials, in a sector of the flash. That sector is kept between flashings to allow for a quick startup after update. On boot, before your sketch is executed, the SDK will retrieve the wifi credentials and attempt a quick connection.
If you change the purpose of your ESP by changing to a completely different sketch, you should wipe the flash with the esptool, and only then flash the new sketch.
Alternatively, there is an ESP.eraseConfig() call that I understand can manually wipe the relevant flash area.
Closing due to non-issue.

closed 2017-10-04 by devyte