esp8266/Arduino #2845: repeatable connection failure with AWS API

bantaj, 2017-01-08:
#### Hardware
Hardware:			ESP-12
Core Version:      	2.0.0

### Description

I am getting repeatable connection failures when trying to make get or post requests to AWS API here: https://fptvt4j3ek.execute-api.us-west-2.amazonaws.com/Temp/mydemoresource

I am able to make similar get requests from other (non-AWS) servers that are also using TLS 1.2 according to their certificate details so it seems this may be an issue with the ESP8266 arduino core.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL

### Sketch

```cpp

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include "WiFiManager.h"          //https://github.com/tzapu/WiFiManager

#ifdef DEBUG_ESP_PORT
#define DEBUG_MSG(...) DEBUG_ESP_PORT.printf( __VA_ARGS__ )
#else
#define DEBUG_MSG(...) 
#endif

// AWS API host and endpoint information:
const char* host = "https://fptvt4j3ek.execute-api.us-west-2.amazonaws.com";
const char* APIURL = "/Temp/mydemoresource";
//const char* host = "www.google.com";
//const char* APIURL = "";

// SSL Certificate finngerprint for the host
const char* fingerprint = "9F 7B 7F 29 FA 8E 4E B7 A8 0A 64 91 74 A6 8A AB 85 6B F4 C0";
//const char* fingerprint = "FF A3 13 31 11 77 3A 70 3F B4 EF 06 4E BF 3B 4E E4 46 F7 FD";

void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
}

void setup() {
  Serial.begin(115200);

  //WiFiManager:
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;
  //reset settings - for testing
  //wifiManager.resetSettings();

  //set callback that gets called when connecting to previous WiFi fails, and enters Access Point mode
  wifiManager.setAPCallback(configModeCallback);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if(!wifiManager.autoConnect()) {
    Serial.println("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(1000);
  } 

  //if you get here you have connected to the WiFi
  Serial.println("connected to wifi! :)");

  delay(3000);
  DEBUG_MSG("bootup...\n");

}

void loop() {
  DEBUG_MSG("loop %d\n", millis());
  delay(1000);  Serial.print("connecting to ");
  Serial.println(host);

  WiFiClientSecure client;
  const int httpsPort = 443;

  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }
  
  if (client.connect(host, httpsPort)) {

  // verify the signature of the ssl certificate
  if (client.verify(fingerprint, host)) {
  Serial.println("ssl cert matches");
  } else {
  Serial.println("ssl cert mismatch");
  }

   // Make an HTTP GET request
  client.println("GET HTTP/1.1");
  client.print("Host: ");
  client.println(host);
  client.println("Connection: close");
  client.println();
   
 delay(500);
 
 // Read all the lines of the reply from server and print them to Serial
 while(client.available()){
 String line = client.readStringUntil('\r');
 Serial.print(line);
 }
 
 Serial.println();
 Serial.println("closing connection"); 
 }
 
 delay(15000);

}
```

Here's the serial output showing the connection failure:
```
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
wifi evt: 0
wifi evt: 3
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.135
connected to wifi! :)
bootup...
loop 6630
connecting to https://fptvt4j3ek.execute-api.us-west-2.amazonaws.com
[hostByName] request IP for: https://fptvt4j3ek.execute-api.us-west-2.amazonaws.com
pm open,type:2 0
[hostByName] Host: https://fptvt4j3ek.execute-api.us-west-2.amazonaws.com lookup error: -5!
connection failed

```

Humancell, 2017-01-08:
Have you tried the `hostname` without the `https://`?

Try using just: `fptvt4j3ek.execute-api.us-west-2.amazonaws.com`

I know that I never put the protocol prefix on hostnames ... just a thought?

bantaj, 2017-01-09:
> Have you tried the hostname without the https://?

@Humancell - Good eye but that's not the issue.  I added the prefix as one of my last attempts at getting it to work but I didn't strip it out before posting the example sketch.

Thanks for catching this.  I'll fix the sample sketch.

liquidfalcon, 2017-01-09:
For what it's worth, there was this commit in the axTLS repository which may be of interest: https://github.com/igrr/axtls-8266/commit/b20140f86747e0570770e78a6fec220c7b77e1d0

ryonlabaw, 2017-01-17:
@bantaj I was able to get a reliable connection to AWS API Gateway + Lambda using the wifi client.

Seems to be issues with other versions of this ESP8266 Core for Arduino ... I was having problems with the git downloaded version.
Works with Core Version: 2.3.0  
Arduino IDE: 1.6.11

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
In addition, there have been several fixes related to connection stability that were merged post 2.4.1.
Closing.

closed 2018-05-29 by devyte