esp8266/Arduino #2847: Processor crash when reading a SPIFFS file if it is closed before end is reached

Bodmer, 2017-01-08:
### Basic Infos
Opening a file in SPIFFS for a read works fine, but if the file is closed before the end is reached by reading, then the processor crashes and tried to reboot.
#### Hardware
Hardware:	NodeMCU 1.0 (ESP-12E)
Core Version:      	2.2.0 and 2.3.0 , other versions not tested

### Description

If a file contains say 150kbytes and it is opened for reading then everything works providing the file is read right to the end. If an attempt to close the file before the end is reached the processor reboots.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:		80MHz and 160MHz
Flash Mode:			Default?
Flash Frequency:	Default?
Upload Using:		Serial
Reset Method:		nodemcu

 Open file, read some bytes, but not all, close file and it crashes

 Open file, read all bytes in file, close file and it continues to work fine.

devyte, 2017-10-04:
@Bodmer is this issue still valid with latest git? If so, please provide an MCVE that shows your observed behavior.

Bodmer, 2017-10-20:
No longer an issue. Thanks.

closed 2017-10-20 by Bodmer