esp8266/Arduino #2853: SPIFFS write not persisting completely on REBOOT

chakri1vr1, 2017-01-12:
### Basic Infos
I am saving some data to a file on SPIFFS in a loop, approximately a line each 25 seconds. When I restart the device, some 3-4 lines at the end which had written are gone on start. As I have read, it seems the op buffers aren't flushed sometimes and sudden reboot might cause similar issue, but my question is that is there a efficient way to handle it? Is it advisable to use fclose() on every write operation (once 25 seconds)

#### Hardware
Hardware:			ESP-12
Core Version:      	2.1.0-rc2

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E) Module
Flash Size: 		4M( 3M SPIFFS)
CPU Frequency:		80Mhz
Upload Using:		SERIAL

### Sketch

```cpp

#include <Arduino.h>
#include <stdlib.h> 
#include <FS.h>

#define SERIAL_BAUDRATE                 115200
#define FORMAT_SPIFF_ON_START                 false
#define PRINT_PENDING_DATA_ON_START                 true

void printFile(File f){
    f.seek(0,SeekSet);
    Serial.println("--------------Starting read of dataFile--------------");
    while(f.available()) {
        //Lets read line by line from the file
        String line = f.readStringUntil('\n');
        Serial.println(line);
    }
    Serial.println("--------------Ended read of dataFile--------------");
}

void updataToServer(postData){
    dataFile.println(postData);
    Serial.print("Appended--");Serial.print(postData);Serial.println("--to dataFile");
}

void setup() {
    Serial.begin(SERIAL_BAUDRATE);
    Serial.println();
    Serial.println();

    SPIFFS.begin();
    if(FORMAT_SPIFF_ON_START)
        if(SPIFFS.format())
            Serial.println("Formatting succesfull");
        else
            Serial.println("Formatting failed");
    dataFile= SPIFFS.open("data.txt", "a+");
    if (!dataFile) {
        Serial.println("Opening dataFile failed");
        handleFatalError();
    }else{
        Serial.println("energyData.txt file opened for appending");
        if(PRINT_PENDING_DATA_ON_START){
            printFile(dataFile);
        }
    }
    if(SPIFFS.info(fs_info)){
        Serial.println("Filesystem information :");
        Serial.print("Total bytes : ");Serial.println(fs_info.totalBytes);
        Serial.print("Used bytes : ");Serial.println(fs_info.usedBytes);
    }else{
        Serial.println("Failed to retrieve FSInfo");
    }
}

void loop() {
    static unsigned long lastDataUpdateLoop = millis();
    if((millis() - lastDataUpdateLoop) > DATA_UPDATE_LOOP_TIME){
        lastDataUpdateLoop = millis();
        updateDataToServer("Some random data to save to file");
    }
}
```

### Debug Messages

```
messages here
```

chakri1vr1, 2017-01-12:
Newbie alert !

A simple `flush()` seems to have solved the issue.

```
dataFile.println(postData);
dataFile.flush();
```

closed 2017-01-12 by chakri1vr1

frank240889, 2018-10-03:
Thanks!!!! I had the same problem!!!!!!