esp8266/Arduino #2854: Documentation error in Arduino/doc/Troubleshooting/debugging.md ?

PaulChristopher, 2017-01-12:
The HelpOnDebugging https://github.com/esp8266/Arduino/blame/master/doc/Troubleshooting/debugging.md

gives an example for developpers, how to debug an application, see code snippet line 83 onwards.

The code introduces a define
```
#ifdef DEBUG_ESP_PORT
#define DEBUG_MSG(...) DEBUG_ESP_PORT.printf( __VA_ARGS__ )
#else
#define DEBUG_MSG(...) 
#endif
```

However the setup function initialised the debug serial as follows
```

void setup() {
    Serial.begin(115200);
    DEBUG_MSG("bootup...\n");
    ....
}
```

Shouldn't it be like this?

```
void setup() {
  #ifdef DEBUG_ESP_PORT
    DEBUG_ESP_PORT.begin(74880);  
  #endif
  DEBUG_MSG("bootup...\n");
   ....
}
```
Moreover wouldn't it be better to use baud raute 74880 instead of 115200, since 74880 the default?
I was always wondering about the gibberish between my debug message when using 115200.

davisonja, 2017-01-12:
> Troubleshooting/debugging.md

Probably would be better to not change serial settings if debug is off, so
long as nothing else relies on it.

I'd rather stick with a more standard rate of 115200. 74880 is a bit more
obscure.
It's a nuisance that the ESP ROM sets serial to that to start with (as you
say, it can involve switching rates to see some startup messages) but
115200 is so much more common.
I wonder how often people are actually needing to see the startup messages
and thus find the difference an issue?

igrr, 2017-01-17:
The SDK startup code switches from 74880 to 115200 internally, so if you switch back to 74880, you may still get some gibberish. Some OS/serial adapter driver combinations don't even support 74880 baud.

closed 2017-01-17 by igrr

PaulChristopher, 2017-01-17:
Agreed, but this was not the main point of the bug report. Apparently, there seems to be a DEBUG_ESP_PORT variable which is set by the IDE to Serial or Serial1. Debug messages are therefore output with DEBUG_ESP_PORT.printf. Thus it should be DEBUG_ESP_PORT.begin and not Serial.begin in the example code, shouldn't it?

igrr, 2017-01-17:
The typical case is that sketch being debugged is already using Serial or Serial1. So the debug port in the menu needs to be set accordingly. Doing this the other way around is possible but seems pretty odd.