esp8266/Arduino #2857: ESP.deepSleep

MisterBE2, 2017-01-14:
### Basic Infos

Hello everyone!

I had tried to put ESP to sleep with ESP.deepSleep() command. And it works, but when it should wake up it just keep restarting. I connected pin 16 (ESP 12-E) to rst...

markusschweitzer, 2017-01-14:
ESP.deepSleep() should start with setup() again (It is like a "fresh" start).

MisterBE2, 2017-01-19:
@markusschweitzer  Can you give me sample code? Because I applied your advice, and IDE tells me: "no matching function for call to 'EspClass::deepSleep()"

duncan-a, 2017-01-19:
What Marcus was saying, is that when the ESP wakes from deep sleep, itis
exactly the same as a reset - the code begins running again from the
beginning.

What you were seeing is perfectly normal, and the only option.

MisterBE2, 2017-01-19:
I feel like I should explain more :/

I'm using ESP12E (nodemcu v3), I connected pin 016 (wake) to rst. And I connected 10K pullup resistor to rst.

Then in void loop() I placed command ESP.deepSleep(10 * 1000000); Till this point everything works fine, but! When this 10 s delay elapse esp keeps restarting in loop. Built in LED is keeps blinking in 0.2 s intervals. 
Wasn't it supposed to restart once (till nex deep sleep command)?
How to fix it?

MisterBE2, 2017-01-20:
I found the solution. You can't have open serial port connection.

jaky000000, 2017-03-10:
overflow
把int改为unsigned int

devyte, 2017-10-04:
@MisterBE2 last comment found a solution => closing as resolved.

closed 2017-10-04 by devyte

didietaja, 2017-12-17:
if you are going to use nodemcu in deepsleep mode, then you cannot use internal LED  in the same time, because they are using the same PIN, on D0/GPIO16.

dan2600, 2018-03-23:
Nice catch, was having the same issue.