esp8266/Arduino #2858: SPIFFS filename length limit?

JamesGKent, 2017-01-14:
### Basic Infos

#### Hardware
Hardware:	Wemos D1 R2
Core Version:      git master

### Description
I'm using the latest git version of this library through the arduino 1.6.9 IDE and I've noticed that one of the files  i loaded into the SPIFFS filesystem was not being served when it was requested. when i tried to debug this i found that SPIFFS.exists(filename) was returning false for this particular file even though it was definitely there. however another file with a similar name (3 chars shorter) was being returned just fine, so i renamed the file to shorten the filename and re uploaded it, at which point SPIFFS.exists succeeds.
the filename in question was this:  
iframeResizer.contentWindow.min.js  
and when shortened:  
iframeResizer.content.min.js  

however another file:
iframeResizer.contentWindow.map  
was being returned just fine.

does the SPIFFS filesystem have some inherent limit on the length of filenames?
if so does this include the whole directory (if it was in a subdirectory) or just the filename?
and if so, what is the limit and why?

vicnevicne, 2017-01-14:
Yes, as indicated in the [readme included with SPIFFS](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/spiffs/README.md), it is designed for small systems with sparse RAM (which is the limited factor on ESP8266). Consequently, it has several simplifications and limitations.
Basically, SPIFFS does not support directories. It's just a list of files. 
But contrary to traditional filesystems, the slash (/) is allowed in filenames, so the code mimics listing files in directory "abc" by listing all filenames that begin with "abc/".
Plus, as you have discovered, there is a total limit of 32 chars for filenames (full path, including directories). I even think (tbc) that the 32 include a mandatory '\0' termination char, so that leaves you with 31 usable characters. 
Consequently, it is advised to keep filenames short and not use deep nested directories.
So basically that's not a bug but a known limitation of SPIFFS. I learned it the hard way too and I think it should be added to the [Filesystem documentation](https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md) ...

JamesGKent, 2017-01-14:
i figured that would probably be the reason, but having a quick look in the docs and the code and not seeing anything jump out at me i figured i would ask the question...
having it added to the documentation would make a lot of sense.

vicnevicne, 2017-01-14:
I just proposed an update to the documentation indeed: https://github.com/esp8266/Arduino/pull/2860 . Hopefully it will be integrated in a future release, if any...
In the sources, the maximum length is defined by [SPIFFS_OBJ_NAME_LEN](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/spiffs/spiffs_config.h#L125) if I'm not mistaken.

closed 2017-02-09 by JamesGKent