esp8266/Arduino #2859: ESP8266 doesnt wake up after put to deepsleep.

thorburn1, 2017-01-14:
Problem description
So, I hooked up my esp8266 with a USB to TTL for 3.3v, connected to the laptop and executed the below code.also connected a reed sensor to it on GPIO pin 2.(if detect changes then it should post data to the server)
Now, it's supposed to POST data to a server, which it's doing fine, but after the first post, it goes to deep sleep(as called by gotosleep function) and never wakes up to detect reed output changes.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k (64k spiffs)
CPU Frequency:      80Mhz
Flash Mode:	        DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck 

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266HTTPClient.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h> 
#include "gpio.h"
extern "C" {
#include "user_interface.h"
  uint16 readvdd33(void);
  bool wifi_set_sleep_type(sleep_type_t);
  sleep_type_t wifi_get_sleep_type(void);
}

// Server details
const char* host = "www.fusmin.com";
const int httpPort = 443;
WiFiClient client;
WiFiManager wifiManager;

// Device ID and Status
byte mac[6];
String MAC,m1,m2,m3,m4,m5,m6,m7;
int status;
int current_state = 0;
int last_state = 0;

void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  Serial.println(myWiFiManager->getConfigPortalSSID());
}

void goingToSleep() {
  yield();
  Serial.println("going to deepsleep");
  delay(100);
  ESP.deepSleep(0);
  yield();
}

void connect_to_wifi() {
  Serial.println("EnterWIFI");
  
   wifiManager.setConfigPortalTimeout(600);
   wifiManager.setAPCallback(configModeCallback);
     if (!wifiManager.autoConnect("latch" , "fusmin123")) {
    Serial.println("failed to connect and hit timeout");
 
 
     }
     Serial.println("connected to wifi :)");
     Serial.println("local ip");
     Serial.println(WiFi.localIP());
  }

void get_IP_and_MAC()
{
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  WiFi.macAddress(mac);
  Serial.print("MAC: ");
  m1 = String(mac[5],HEX);
  m2 = String(mac[4],HEX);
  m3 = String(mac[3],HEX);
  m4 = String(mac[2],HEX);
  m5 = String(mac[1],HEX);
  m6 = String(mac[0],HEX);
  MAC = m1+m2+m3+m4+m5+m6;
  Serial.println(MAC);
}
// Initial Setup - One Time

void setup() {
  Serial.begin(115200);
  pinMode(2, INPUT);
  MAC = String();
  m1 = String();
  m2 = String();
  m3 = String();
  m4 = String();
  m5 = String();
  m6 = String();
  //ESP.reset();
  connect_to_wifi();
  get_IP_and_MAC();

} 

void loop() {

   if (digitalRead(2) == HIGH) {
    current_state = 1;
  }
  else {
    current_state = 0;
  }

  if (current_state == last_state) {
  
    goingToSleep();
  }
  
 else
    {
   connect_to_wifi(); 
   get_IP_and_MAC();
    }
  Serial.print("connecting to ");
   Serial.println(host);
   
   if (!client.connect(host, httpPort)) 
   {
    Serial.println("connection failed");
    return;
   } 
   else 
   {

      post_data_server();
      while (client.available()) {
        String line = client.readStringUntil('\r');
        Serial.print(line);
        Serial.println();
        Serial.println("closing connection");
      }

    }
    
    
  }

void post_data_server() { 
  HTTPClient http;
  http.begin("https://www.fusmin.com/mlatch/logapi.php" ,  "F6 55 EE 84 54 DA CF 23 82 90 F4 BA D2 F5 30 AA FC B8 D3 7A‎");
  http.addHeader("Content-Type", "application/json");
  String postMessage = "{\"device_id\":\"";
  postMessage += MAC;
  postMessage += "\", \"status\":\"";
  postMessage += current_state;
  postMessage += "\"}";
  int httpCode = http.POST(postMessage);
  Serial.print("HTTP Result:");
  Serial.println(httpCode);
  http.writeToStream(&Serial);
  String payload = http.getString();
  http.end();
  last_state = current_state;
}

vlast3k, 2017-01-14:
GPIO2 should be floating or pulled up during start... if your reed sensor
is keeping it low, this may be the reason

thorburn1, 2017-01-14:
I used a 10k resistor for that between , i forgot to mention , Still doesnt comes out of wake up.

vlast3k, 2017-01-14:
and what does deepSleep(0) stand for... it should be rather the time in
deepSleep in us
and it is good to add a 1-2 seconds delay after the deepSleep, as the ESP
needs some time to go into deepSleep

duncan-a, 2017-01-14:
I only had a quick look, but I can't see the 'current state' value being
transferred to 'last state' anywhere...

thorburn1, 2017-01-14:
Its in the post_data_server function , last line.

thorburn1, 2017-01-14:
vlast3k :ESP.deepSleep(10000);
               delay(3000);
Doesn't help.

vlast3k, 2017-01-15:
aaahh i missed a very important point
you need to connect GPIO16 to RST
when the device is about to wake up, it will trigger rst low/high and the
device reboots and execution continues

closed 2017-01-15 by thorburn1