esp8266/Arduino #2861: HTTP client file download.

fredericplante, 2017-01-14:
Is there an example that use the HTTP client to connect  to a HTTP server, request a file, download it and then save it to SPIFFS?

You know, what a HTTP client should be doing. :D

vicnevicne, 2017-01-14:
:-)
Please note that for usability questions, you'll have much more attention on the esp8266.com forums. Issues are meant to report bugs or problems in the project itself.
That being said, a few  once worked on an http proxy on the ESP that uses SPIFFS for temporary storage, so it does what you request, plus more. 
The code is at http://www.esp8266.com/viewtopic.php?f=32&t=10103&start=16#p48617

devyte, 2017-10-04:
@fredericplante there is an httpClient class implementation. But this is not the right place to ask for usage help.
Closing due to off-topic, see #3655 .

closed 2017-10-04 by devyte

Erol444, 2018-09-07:
I got it working like this :) you give him file url and program saves the file to SPIFFS. 
```
 url = String(UPDATE_URL) + response.substring(4);
    //url = http://123.123.123.123/SPIFFS/test2.htm
    String file_name=response.substring(response.lastIndexOf('/'));
    //file_name = test2.htm
    Serial.println(url);
    File f = SPIFFS.open(file_name, "w");
    if (f) {
      http.begin(url);
      int httpCode = http.GET();
      if (httpCode > 0) {
        if (httpCode == HTTP_CODE_OK) {
          http.writeToStream(&f);
        }
      } else {
        Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
      }
      f.close();
    }
    http.end();
```

KushlaVR, 2018-12-11:
http - what is it? what type? Where did you copy this pice of code?

Erol444, 2018-12-11:
> http - what is it? what type? Where did you copy this pice of code?

If you want to download A file(any type I assume) and save it to SPIFFS you can use the code above. This code should be copied to your ESP8266 program.

KushlaVR, 2018-12-12:
> > http - what is it? what type? Where did you copy this pice of code?
> 
> If you want to download A file(any type I assume) and save it to SPIFFS you can use the code above. This code should be copied to your ESP8266 program.

To make this code work you'll nead:
1. #include <ESP8266HTTPClient.h>
2. declare http variable
HTTPClient http;

rtek1000, 2018-12-13:
Hello,

Since it is possible to download,

Would it be possible to measure the internet speed (up to the bandwidth that the ESP8266 supports)?

Would it be possible to download the data without saving to be able to download a larger file, maybe a 5MB or 10MB to perform an average speed?

Any suggestion?

KushlaVR, 2018-12-13:
If you commentout **f** variable - then you can read data without saving.
Or if uou use **Erol444** example (do download http:// files) - you'll nead to write empty class based on Stram
Then you can cound number of bytes recived and timespan.

Erol444, 2018-12-13:
@rtek1000 Well actually **HTTPClient::writeToStream** function(the one I used in example code above) **returns bytes written** (negative values are error codes ).

As for not saving the file, it's been quite some time since I last programmed esp32 (I don't have one to test anymore) but you should also check in **HttpClient** class, there is a function **getSize**, maybe you can get it working:

> 
>   //size of message body / payload
>   //@return -1 if no info or > 0 when Content-Length is set by server
> 
> int HTTPClient::getSize(void)
> {
>     return _size;
> }

fredericplante, 2020-04-21:
It is nice to see that this turn out to be usefull after all.

Thank you all for the suggestion, I see it could also be easily modified to save to a SDcard, that is nice.

:)