esp8266/Arduino #2865: HTTPS call fails

pkolmann, 2017-01-15:
### Basic Infos

#### Hardware
Hardware:			Wemos D1 mini
Core Version:      	2.3.0

### Description

I want to add Generic HTTPS to ESPeasy and stumbled accross the problem, that I always get the error 'please start sntp first !'. I googled this error and came accross issue #1375. I tested the cottak's sketch (file ssl-test.ino) 

with my domain and I got the following stacktrace:

0x4010020c: _umm_free at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010020c: _umm_free at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x40203758: ax_port_free at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 609
0x4022144a: ssl_free at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 254
0x40203170: SSLContext::_readAll() at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
0x40203319: SSLContext::available() at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
0x40203367: WiFiClientSecure::connected() at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
0x40205d32: HTTPClient::connected() at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 245
0x402043fe: HTTPClient::writeToStream(Stream*) at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 197
0x40204e28: String::changeBuffer(unsigned int) at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40204574: HTTPClient::getString() at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 197
0x40204aa5: Print::write(char const*) at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x40204ad0: Print::println() at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x40202324: loop at D:\Philipp\ESP\ssl-test/ssl-test.ino line 42
0x402059c4: loop_wrapper at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\birgit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

With the DS18B20_Temp.ino sketch on the same board with the same host it works with https.

### Settings in IDE

Module:  			WeMos D1 R2 & mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL
Reset Method:		

### Sketch
see attached zip file

[ESP-Bug.zip](https://github.com/esp8266/Arduino/files/706574/ESP-Bug.zip)

pwrsoft, 2017-09-19:
I can replicate this as well. Any updates with the issue?

devyte, 2018-07-13:
Closing due to merged bearssl (axtls will be deprecated).

closed 2018-07-13 by devyte