esp8266/Arduino #2866: Loop in loop() causing Soft WDT reset

ludekvodicka, 2017-01-15:
### Basic Infos

#### Hardware
Hardware:			WemosD1 mini v 2.1.0

### Description

I'm getting "Soft WDT reset" in simple loop.

```
Soft WDT reset

ctx: cont 
sp: 3ffef1d0 end: 3ffef3d0 offset: 01b0

```

Stack (I'm not able to decode it. i don't have such option in my arduino app)

```
>>>stack>>>
3ffef380:  3ffe8430 feefeffe 3ffef360 3ffef390  
3ffef390:  40106af0 00000338 00000000 3ffee39c  
3ffef3a0:  3fffdad0 00000000 00000338 40201c55  
3ffef3b0:  feefeffe feefeffe 3ffee394 40202138  
3ffef3c0:  feefeffe feefeffe 3ffee3b0 40100114  
<<<stack<<<

```

This is sample code, It's very simplifed version of my app only to present the bug. App crash always when counter is 824.

```
void setup()
{
  Serial.begin(9600);
  Serial.println("Start");
  pinMode(D1, OUTPUT);
}

void loop()
{
  for ( long nBlinkLong=0; nBlinkLong < 20 * 1000; nBlinkLong++ )
  {
    Serial.printf("%d,", nBlinkLong);
    digitalWrite(D1, LOW);
    delayMicroseconds(1000);
  }
}
```

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz

ludekvodicka, 2017-01-15:
Ok, it seems that it's caused by watchdog timer. I didn't know about this before.

https://github.com/esp8266/Arduino/issues/34#issuecomment-91751313

JamesGKent, 2017-01-16:
could you try changing your code to use delay instead of delayMicroseconds? using delay allows some of the background tasks to happen during the delay which may prevent the watchdog reset (not tested this) where delayMicroseconds does not.

ludekvodicka, 2017-02-09:
Hi James, sorry for late reply. I tried to use delay but with same results. The second problem was that I needed delayMicroseconds because of simulating PWM. But I solved it without delay/delayMicrosends with getTickCount measurement.

JamesGKent, 2017-02-09:
glad you solved your problem, can i suggest that you post a stripped down example of your fixed code (in case someone else has a similar issue) and then close this issue?

ludekvodicka, 2017-02-09:
Sure, it's something like this

```
unsigned long timeLastCheck = 0;
unsigned long intervalCheck = 500;

void loop() {
  unsigned long timeNow = millis();
  if ( timeLastCheck == 0 || timeNow-timeLastCheck > intervalCheck )
  { 
    ...
  }
}
```

closed 2017-02-09 by ludekvodicka

hemangjoshi37a, 2017-06-06:
can you paste the whole code please?? I want to see where exactly the code goes and how to use it.
Thax.

regimantas, 2017-11-15:
Same problem with HX711 library on esp8266. delay(0); in loop fix problem :) Library use yield(); but it not work :(

Ginger-Stone, 2018-05-18:
@regimantas did you manage to fix your problem with the hx711?

doublejosh, 2020-02-23:
I found this was necessary even in a particularly long running `for` loop, rather than just longer running `while` statements. Using `delay(0)` worked perfectly. Thanks!

Deepakg1105, 2020-05-02:
> I found this was necessary even in a particularly long running for loop, rather than just longer running while statements. Using delay(0) worked perfectly. Thanks!

But delay() has some issues like blocking other connected devices for specified time and non-accurate delays as code takes its own time to run.

dungcnttimd, 2021-11-19:
I made this mistake. Sorry can anyone help me?

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 01a0
3fffff70:  5301a8c0 00ffffff 0101a8c0 40201bfc  
3fffff80:  3ffe85cc 3ffee508 3ffee5b8 4020105c  
3fffff90:  40204dfc 5301a8c0 feefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffee5b8 402022d4  
3fffffb0:  feefeffe feefeffe 3ffe85e0 40100c01  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00042d10
~ld

My code

 Serial.begin(115200);

  Serial.printf("Connecting to %s \n", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED){
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.println(WiFi.localIP());