esp8266/Arduino #2868: 'class EEPROMClass' has no member named 'length'

SkazochnikZlodey, 2017-01-15:
Hello after adding bord by board manager from http://arduino.esp8266.com/stable/package_esp8266com_index.json

#### Hardware
Hardware:			Wemos D1 mini pro
### Description

Problem description
Hello after adding bord by board manager from http://arduino.esp8266.com/stable/package_esp8266com_index.json
got error : 'class EEPROMClass' has no member named 'length' 
before compilation was successful. 
Looks like EEPROM.h is old

### Settings in IDE

Module:  			Wemos D1 R2 & mini
Flash Size: 		4MB/3MB SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		COM8 SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch
 /*
 * EEPROM Read
 *
 * Reads the value of each byte of the EEPROM and prints it
 * to the computer.
 * This example code is in the public domain.
 */

#include <EEPROM.h>

// start reading from the first byte (address 0) of the EEPROM
int address = 0;
byte value;

void setup() {
  // initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
}

void loop() {
  // read a byte from the current address of the EEPROM
  value = EEPROM.read(address);

  Serial.print(address);
  Serial.print("\t");
  Serial.print(value, DEC);
  Serial.println();

  /***
    Advance to the next address, when at the end restart at the beginning.

    Larger AVR processors have larger EEPROM sizes, E.g:
    - Arduno Duemilanove: 512b EEPROM storage.
    - Arduino Uno:        1kb EEPROM storage.
    - Arduino Mega:       4kb EEPROM storage.

    Rather than hard-coding the length, you should use the pre-provided length function.
    This will make your code portable to all AVR processors.
  ***/
  address = address + 1;
  if (address == EEPROM.length()) {
    address = 0;
  }

  /***
    As the EEPROM sizes are powers of two, wrapping (preventing overflow) of an
    EEPROM address is also doable by a bitwise and of the length - 1.

    ++address &= EEPROM.length() - 1;
  ***/

  delay(500);
}

yar-trach, 2017-07-31:
I also have this issue. NodeMCU 1.0 on 8266

1330154737, 2017-08-31:
I  have this issue. NodeMCU 1.0 on 8266,too

MiguelNaumSantana, 2017-09-06:
Also here!

devyte, 2017-10-04:
@SkazochnikZlodey A ::length() method implementation that just returns _size should be enough. 
Care to make a PR?

devyte, 2017-12-05:
Fixed via #3855 . Closing.

closed 2017-12-05 by devyte