esp8266/Arduino #2870: Support explicitly call GC during idle times on SPIFFS

joelucid, 2017-01-16:
SPIFFS writes get very slow when there are no free blocks available. Therefore it makes sense to clear blocks in the background when the application is idle. The following patch exposes a gc() call which in turn calls SPIFFS_gc_quick which serves exactly this purpose.

diff --git a/cores/esp8266/FS.cpp b/cores/esp8266/FS.cpp
index 6ae11e1..6bd359c 100644
--- a/cores/esp8266/FS.cpp
+++ b/cores/esp8266/FS.cpp
@@ -173,6 +173,13 @@ void FS::end() {
     }
 }
 
+bool FS::gc() {
+    if (!_impl) {
+        return false;
+    }
+    return _impl->gc();
+}
+
 bool FS::format() {
     if (!_impl) {
         return false;
diff --git a/cores/esp8266/FS.h b/cores/esp8266/FS.h
index 79620f9..ce42ccf 100644
--- a/cores/esp8266/FS.h
+++ b/cores/esp8266/FS.h
@@ -125,6 +125,7 @@ public:
     bool rename(const char* pathFrom, const char* pathTo);
     bool rename(const String& pathFrom, const String& pathTo);
 
+	bool gc();
 protected:
     FSImplPtr _impl;
 };
diff --git a/cores/esp8266/FSImpl.h b/cores/esp8266/FSImpl.h
index e5694b5..242811c 100644
--- a/cores/esp8266/FSImpl.h
+++ b/cores/esp8266/FSImpl.h
@@ -71,6 +71,8 @@ public:
     virtual DirImplPtr openDir(const char* path) = 0;
     virtual bool rename(const char* pathFrom, const char* pathTo) = 0;
     virtual bool remove(const char* path) = 0;
+	virtual bool gc() = 0;
+	
 
 };
 
 #if !defined(NO_GLOBAL_INSTANCES) && !defined(NO_GLOBAL_SPIFFS)
diff --git a/cores/esp8266/spiffs_api.h b/cores/esp8266/spiffs_api.h
index f98df11..f42e43e 100644
--- a/cores/esp8266/spiffs_api.h
+++ b/cores/esp8266/spiffs_api.h
@@ -165,6 +165,12 @@ public:
         return true;
     }
 
+	bool gc() override 	{
+		return SPIFFS_gc_quick( &_fs, 0 ) == SPIFFS_OK;
+	}
+	
+
+

devyte, 2017-10-04:
@joelucid this looks interesting. Care to make a PR?

TD-er, 2019-04-01:
What is the status of the internal garbage collector right now?
I've only seen it used when there is some hardly any space left.
https://github.com/esp8266/Arduino/blob/f706c83b66261df58b7d6148e343fd21cbd77206/cores/esp8266/spiffs/spiffs_nucleus.cpp#L449-L468

Is there a better way to help improve SPIFFS performance? Or is this currently the best way to do it?
If so, then I can make a PR for it, since it looks like a good command to have and perhaps call it when deleting a (large) file.

devyte, 2019-04-02:
No work has been on this front. As you said, this applies in the case where there is little space left, hence the smaller SPIFFS sizes.

TD-er, 2019-04-03:
It looks to me, this `spiffs_gc_quick` is only called from within the write block.
But this garbage collection cannot do garbage collection when there are open file descriptor handles with something other than read-only.
So this means it is not really cleaning when writing.

This does make it even more useful for a separate GC function to be called.
Also it would make sense to call it ourselves right after mounting the file system and before opening any file, and in tasks where files are to be written.

I will create a PR with the changes suggested in the opening post.

earlephilhower, 2019-04-10:
Closed with #5944

closed 2019-04-10 by earlephilhower