esp8266/Arduino #2872: wificlient doesn't remove chunk size from HTTP GET response body

electronicsguy, 2017-01-16:
### Basic Infos

#### Hardware
Hardware: ESP-01
Core Version:      	?2.1.0-rc2?

### Description

I'm using wificlientsecure (derived from wificlient) to fetch remote data. While the basic functionality itself is flawless, the body of the HTTP GET response contains chunk size before every chunk. Something like this is returned:

```
5
Hello
0

```

Of course, I could ignore the first line before printing out the response. But why isn't the chunk size removed by the library handling the response? 

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

cimba007, 2017-02-04:
Got the same problem on 2.3.0 *bump*

cimba007, 2017-02-04:
From my understanding the following is happening:

Using this piece of code everything is working as expected:

```
// file found at server
if(httpCode == HTTP_CODE_OK) {
  http.writeToStream(&USE_SERIAL);
```

It has the drawback that we can't process the code in "between".

On the other hand _**StreamHTTPClient.ino**_ gives us a nice example on how to to streaming stuff .. with the drawback of the "left over" chunk size. As the first function writeToStream is processing the whole stuff in once piece it can read the "chunk" size .. and process only chunk_size bytes at once .. leaving it with the capability to just remove the "size" in the beginning.

But what happens if we can only look at smaller pieces as shown in the streamhttpclient? e.g. the chunk size is 2000bytes and the buffer size is 128bytes .. I don't know how to find out where the chunk-size stuff is written? It might be just in the 128bytes buffer, or even **BETWEEN** two 128byte-buffers.

Only chance I see is to make the buffer big-enough to hold a whole chunk!

Then use this piece of code from writeToStream:

> // read size of chunk
> len = (uint32_t) strtol((const char *) chunkHeader.c_str(), NULL, 16);
> size += len;
>  DEBUG_HTTPCLIENT("[HTTP-Client] read chunk len: %d\n", len);

in combination with 

> size_t size = stream->available();
> if(size) {
>    // read up to 128 byte
>    int c = stream->readBytes(buff,   **((size > sizeof(buff)) ? sizeof(buff) : size)**  );

But modify the highlighted size-piece to only read chunk-size

With this we know that after chunk-size is written to the buffer the next "// read size of chunk" will succeed.

electronicsguy, 2017-04-03:
Ok, so this was not an error, but a lack of understanding on my part. wificlientsecure works at the transport layer(?), not at the application/presentation layer. So it is correctly returning chunk sizes for cases when the transfer-encoding is "chunked". We have to use a HTTP client or write one ourselves to correctly parse chunked responses. I just wrote my own mini HTTP client to work with wificlientsecure.

closed 2017-04-03 by electronicsguy