esp8266/Arduino #2873: Httpupdate is compatible with OTA ?

odilonafonso, 2017-01-16:
### Basic Infos

#### Hardware
Hardware:		Wemos D1-R1 mini
Core Version:      	2.3.0-rc2

### Description

Problem description

### Settings in IDE

Module:  			Wemos D1-R1 mini
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Upload Using:		OTA / SERIAL
Reset Method:		nodemcu

Can I use ArduinoOTA and Httpupdate together ?

I try to upload a sketch with OTA but get:

[ERROR]: No response form device.

My sketch use ArduinoOTA and Httpupdate code. Httpupdate is workink good, I can upload any bin file I want from internet, but OTA does not, in my local net.
Can someone please help me?

odilonafonso, 2017-02-10:
Hi  @krzychb, 

Can you help me with this?
OTA on my local network port worked fine, but now it does not work anymore.

I always get:

> [ERROR]: No device response

And I noticed that the ESP does an auto-reset.

closed 2017-04-05 by odilonafonso

gryzli133, 2021-01-29:
Did you find solution? I have the same problem, for last month I did hundreds OTA uploads, but since few days not even one is possible, giving similar warnings like you described. Did you find a way to combine OTA and httpupload in one sketch?