esp8266/Arduino #2876: ESP8266httpUpdate

neilyoung, 2017-01-17:
I'm using a NodeMCU v2 board with Atom IDE and platform.io. The update sketch is simply exactly the example sketch available in ESP8266httpUpdate/examples/httpUpdate/httpUpdate.ino.

The symptom is, that the very first fw update after flashing the sketch does not recover from deep sleep.

This is the console print then:

```
[SETUP] WAIT 2...
[SETUP] WAIT 1...
sleep disable
pm open,type:0 0

state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Then nothing else happens. If I'm hard-resetting the device via button, the _not_updated_ script starts and runs like hell forever then. Seems, that the very first fw flash after Atom's flashing fails.

The sequence above looks like this then (in every round):

```
sleep disable
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
@cp:0
ld
```
.. and it restarts properly.

Known problem? Can somebody confirm?

sglvladi, 2017-01-29:
Hi @neilyoung , 

Referring to this closed issue: [https://github.com/esp8266/Arduino/issues/1722], it appears that this "problem is known and only happens at the first restart after serial flashing. if you do one manual reboot by power or RST pin all will work".

Thus, it is not related to the ESP8266httpUpdate library, but it's rather an inherent "bug" of the esp. I am putting the quotes around "bug", because if you perform a single reset after every upload, which is generally advised anyways, you should never come across this error ever again. :)

Regards.

neilyoung, 2017-01-29:
Thanks, but this is not a solution for me. I was looking for a way to automatically update the firmware. Having to press a reset button (which I do no have) is not a solution.

sglvladi, 2017-01-29:
Hi again,

Maybe I should have been more specific. You shouldn't need to reset after every upload, but after every **Serial** upload (i.e. from cable), which means that, in my experience, consecutive resets after http updates should work just fine! Finally, in your case, since you don't have a button, it suffices to pull RST to ground.

Hope this helps.

Regards

neilyoung, 2017-01-29:
OK, to summarize the problem and the current situation:

1) I'm flashing a  new FW via USB. Doing no kind of reset afterwards
2) I'm doing a FW update OTA. The final RESET leaves the device hanging in a Panic, WDT reset and all this -> doesn't work

What I have successfully tested is this:
1) I'm flashing a  new FW via USB. 
2) Right before the FW update over OTA I'm doing a short deep sleep (probably a restart would do that as well, need to test)
3) After return from deep sleep I start the FW update over OTA. The final RESET following the successful update goes through without problems and the new FW is there

devyte, 2017-10-04:
@neilyoung did you try this:
- flash over USB
- reset via the physical button
- do normal OTA update, without deepsleep etc.

Please report back here whether the above works for you.

devyte, 2017-12-05:
Closing due to lack of feedback in two months, and age.

closed 2017-12-05 by devyte