esp8266/Arduino #2884: update wire.h from 1.0 to 2.3.2

FischerMartin, 2017-01-19:
Hey Guys awesome work with this project! Is it possible to upgrade/merge the wire.h library? I have some problems with an sensirion i2c sensor. With the new libary everything works fine, but sadly not with the old one.

the problem is in -> Wire\src\utility\twi.c 

we have to upgrade these two files:

#include <avr/io.h>
#include <avr/interrupt.h>

greets 
Martin

devyte, 2017-10-04:
@FischerMartin I do see that the current Wire lib shows 1.0 under library.properties. However, what do those two avr file have to do with the ESP?

FischerMartin, 2017-10-04:
Especially for this case, I need the wire.stop functionality, which is included in the 2.3.2 wire library.
This request is a bit older and I'm not 100% sure why I marked the <avr> scripts. Maybe I got some compiler issue while I'm trying the upgrade on my own way.

greets'

earlephilhower, 2019-07-21:
`Wire` is implemented with a custom library per controller.  And, in fact, on base Arduino it's still 1.0 and has no `::stop` member:
https://github.com/arduino/ArduinoCore-megaavr/blob/master/libraries/Wire/library.properties
https://github.com/arduino/ArduinoCore-megaavr/blob/master/libraries/Wire/src/Wire.h

devyte, 2019-10-09:
The functionality needed here is a Wire::end() method that shuts down the internals and leaves everything in a known state. In other words, the reverse of begin() + reset the object and any global state.
Help wanted.

devyte, 2019-10-29:
CC @Tech-TX

Pushing back.

devyte, 2019-12-18:
Reducing 2.7.0 scope. This is to be handled as part of the Wire ongoing rework.