esp8266/Arduino #2885: ESP.eraseConfig() and ESP.restart() not erase WPS config.

rudacs, 2017-01-19:
### Basic Infos

#### Hardware
Hardware: ESP-12e
Core Version: 2.3.0

### Description

I need to clear the WPS configuration directly by code.
I Tried to do:

ESP.eraseConfig();
ESP.restart();

### Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

void setup() {
Serial.begin(74880);
}

void loop() {
if (Serial.available() > 0)
  {
    String text = Serial.readString();
    text.trim();

    Serial.print("Command: ");
    Serial.println(text);

    if (text == "CLEAR")
    {
      ESP.eraseConfig();
      ESP.restart();
    }
    else if (text == "WPS")
    {
      startWPSPBC();
    }
  }
}
```

frankalicious, 2017-01-20:
Does replacing  
`ESP.restart();`
with 
`ESP.reset();`
work?
See https://github.com/esp8266/Arduino/issues/1494

bestpika, 2017-05-18:
WiFi config not erase.
```cpp
void formatConfig() {
    Serial.println(ESP.eraseConfig());
    delay(5000);
    ESP.reset();
}
```

devyte, 2017-10-04:
@rudacs @bestpika Could you please try PR #3215 ? It has the latest SDK. I'm not aware of any relevant changes, but it's worth a try.

igrr, 2017-10-04:
Also please try https://github.com/esp8266/Arduino/pull/3635.

tablatronix, 2017-10-29:
Only #3635 fixes this for me, I am unable to get eraseConfig to do anything in 1.5.3. FYI

devyte, 2018-01-08:
The referenced PR is already merged, and there is confirmation by @tablatronix , so closing.

closed 2018-01-08 by devyte