esp8266/Arduino #2886: Raising Exception29 on building Wificlient

alemlishar, 2017-01-20:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			esp8266
Core Version:      	?2.1.0-rc2?

### Description
Hello all,

I am basically testing from the example, wificlient, and modified some lines to connect to my local server, and after some times it crushed with displaying the below error log, i appreciate any help, thank you, the code and the log is here below

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp
#include <ESP8266WiFi.h>

const char* ssid     = "****";
const char* password = "******";

const char* host = "******";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup() {
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

int value = 0;

void loop() {
 // delay(5000);
  ++value;

  Serial.print("connecting to ");
  Serial.println(host);
  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 6065;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  
  // We now create a URI for the request
  client.print("hello");
  client.stop();
  }

```

### Debug Messages

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffcd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe70:  3fff017c 00000203 3fff4864 401004f4  
3ffffe80:  00000000 3fff4864 3fff4864 40106ff8  
3ffffe90:  3fffc278 4010268c 3fffc200 40220a78  
3ffffea0:  3ffe0000 00790171 40210017 3ffefdd4  
3ffffeb0:  3ffefcbc 3ffefcc4 00000011 00000001  
3ffffec0:  40000f58 00000000 00000020 3ffe9aee  
3ffffed0:  3fff017c 3fff4864 3ffefcdc 40223944  
3ffffee0:  00000035 00000000 3ffe9ada 3ffefcc4  
3ffffef0:  000000c0 3fffdcc0 3ffe8c80 40106fb4  
3fffff00:  00000000 00000000 3ffe9504 3ffefcc0  
3fffff10:  3ffe9ada 3ffefcc4 3fff4864 40225080  
3fffff20:  3ffef9e4 3fff017c 3fff017c 3ffed740  
3fffff30:  00000000 3fff4864 0000001c 3fff017c  
3fffff40:  3ffe9acc 00000000 3fff4864 40224489  
3fffff50:  012ba8c0 00000078 00000000 00000027  
3fffff60:  00000002 0000001a 40204e9b 3ffec620  
3fffff70:  3ffe9aa4 3fffdcc0 3ffe8c88 3ffe8c88  
3fffff80:  40204e0e 3ffec620 3fffdab0 3fff01cc  
3fffff90:  3fffdc80 00000000 3fff4864 40218f0b  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

alemlishar, 2017-01-22:
Hello all,

i am still waiting stuck with this bug, can anybody help me out, thank you

Regards.

comino, 2017-01-26:
Can you decode the stackdump with 
https://github.com/littleyoda/EspStackTraceDecoder
or: 
https://github.com/me-no-dev/EspExceptionDecoder

alemlishar, 2017-01-28:
hello comino,

i cant manage to decode it, it always bring this stack error

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys
sp: 3ffffcd0 end: 3fffffb0 offset: 01a0

            stack>>>
            3ffffe70: 3fff017c 00000203 3fff4864 401004f4
            3ffffe80: 00000000 3fff4864 3fff4864 40106ff8
            3ffffe90: 3fffc278 4010268c 3fffc200 40220a78
            3ffffea0: 3ffe0000 00790171 40210017 3ffefdd4
            3ffffeb0: 3ffefcbc 3ffefcc4 00000011 00000001
            3ffffec0: 40000f58 00000000 00000020 3ffe9aee
            3ffffed0: 3fff017c 3fff4864 3ffefcdc 40223944
            3ffffee0: 00000035 00000000 3ffe9ada 3ffefcc4
            3ffffef0: 000000c0 3fffdcc0 3ffe8c80 40106fb4
            3fffff00: 00000000 00000000 3ffe9504 3ffefcc0
            3fffff10: 3ffe9ada 3ffefcc4 3fff4864 40225080
            3fffff20: 3ffef9e4 3fff017c 3fff017c 3ffed740
            3fffff30: 00000000 3fff4864 0000001c 3fff017c
            3fffff40: 3ffe9acc 00000000 3fff4864 40224489
            3fffff50: 012ba8c0 00000078 00000000 00000027
            3fffff60: 00000002 0000001a 40204e9b 3ffec620
            3fffff70: 3ffe9aa4 3fffdcc0 3ffe8c88 3ffe8c88
            3fffff80: 40204e0e 3ffec620 3fffdab0 3fff01cc
            3fffff90: 3fffdc80 00000000 3fff4864 40218f0b
            3fffffa0: 40000f49 3fffdab0 3fffdab0 40000f49
            <<

            <stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(1,6)

ets Jan 8 2013,rst cause:4, boot mode:(1,6)

wdt reset

devyte, 2017-10-04:
@alemlishar I think you have 2 problems.
1. there is a known issue where you have to manually press the reset button after flashing over serial, in order to assure that your sketch runs correctly. I say this due to the double reset at the end.
2. You should space out the execution of the loop body. The loop executes continously, as fast as possible, with only servicing the wifi stack in between. As it is, the code is attempting to connect, communicate, and disconnect at top speed. I suggest calculating the elapsed second with millis(), and then execute the body only after e.g.: 1s has elapsed.

In any case, this issue is rather old. Please also try with latest git.

devyte, 2017-11-11:
Closing due to age and no feedback in over a month.

closed 2017-11-11 by devyte