esp8266/Arduino #2887: how to use MD5.h

wony365, 2017-01-20:
there is a MD5.h,but there is not a example 

https://github.com/esp8266/Arduino/tree/master/cores/esp8266 

i used an arduino md5 example like this  https://github.com/tzikis/ArduinoMD5

MD5_Hash.ino

///////////////////////////////////////////////////

#include <MD5.h>
/*
This is en example of how to use my MD5 library. It provides two
easy-to-use methods, one for generating the MD5 hash, and the second
one to generate the hex encoding of the hash, which is frequently used.
*/
void setup()
{
  //initialize serial
  Serial.begin(9600);
  //give it a second
  delay(1000);
  //generate the MD5 hash for our string
  unsigned char* hash=MD5::make_hash("hello world");
  //generate the digest (hex encoding) of our hash
  char *md5str = MD5::make_digest(hash, 16);
  free(hash);
  //print it on our serial monitor
  Serial.println(md5str);
  //Give the Memory back to the System if you run the md5 Hash generation in a loop
  free(md5str);
}

void loop()
{
}
///////////////////////////////////////////////////////////////////

but when i complie it in  generic esp8266 module, it Prompt error message：
'MD5' has not been declared , 

how to use md5.h encoding string ，waitting for reply! thank you very much

bauerj, 2017-01-23:
This is already answered: #1349

devyte, 2017-09-09:
Given the previous comment, I am closing this.

closed 2017-09-09 by devyte

gautham1902, 2019-10-25:
the above code is not working it returns no matching function to call

francwalter, 2022-07-19:
So what is the benefit to use [tzikis/ArduinoMD5](https://github.com/tzikis/ArduinoMD5) if there is already an easier working md5 in the core?