esp8266/Arduino #2888: NodeMCU 0.9 (ESP-12 module)wount solve the problem epscomm_upload_mem_failed

mwaz, 2017-01-21:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

bauerj, 2017-01-23:
You need to give more detail in order to get a helpful response.

Is the ESP8266 even in bootloader mode? See [I am getting "espcomm_sync failed" error when trying to upload my ESP. How to resolve this issue?](https://github.com/esp8266/Arduino/blob/master/doc/faq/a01-espcomm_sync-failed.md) for more details.

devyte, 2017-10-04:
Closing due to lack of info, and probably the title refers to the esp comm error message, which is documented.

closed 2017-10-04 by devyte