esp8266/Arduino #2891: Connecting NodeMcu Lua ESP8266 ESP-12E CH340G DEv board Module with Arduino IDE

RamanujamV, 2017-01-22:
#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

I have a problem using NodeMcu Lua ESP8266 ESP-12E CH340G WIFI Internet Development Board Module with an Arduino IDE 1.8.1 running in a Raspberry Pi 2 Model B
(Rpi OS: "Raspbian GNU/Linux 8 (jessie)")

when I connect the ESP-12 board to the Raspberry Pi USB port, I do not see any additional Port assigned and added - in addition to the existing Serial port "/dev/ttyAMA0"- 
similar to the connection of  an Arduino: I see an additional port ("/dev/ttyACM0") listed. 

Module:  			NodeMcu Lua ESP8266 ESP-12E CH340G WIFI Internet Development Board Module
Flash Size: 		4MB
CPU Frequency:    80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		Raspberry Pi USB port
Reset Method:		nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

vicnevicne, 2017-01-22:
This most probably is not an issue with the Arduino IDE cores for ESP8266.
May I suggest you to close this issue and post it on the user forums at esp8266.com where you will probably get help faster than here.

RamanujamV, 2017-01-22:
Thanks for your prompt response!

closed 2017-01-22 by RamanujamV