esp8266/Arduino #2893: How to use WiFi Event in multiple files project ?

czopas, 2017-01-23:
### Basic Infos

#### Hardware
Hardware:			ESP-12S
Core Version:      	        2.2.0

### Description

I have a problem using WiFi Event on multiple files project. Could you help me and give an advice ?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

#### main.cpp

```cpp

#include <Arduino.h>
#include <http.h>

IotHttpClass HTTP;

void setup() {

  static WiFiEventHandler e1;
 
  e1 = WiFi.onStationModeGotIP(HTTP.onSTAGotIP);

}

void loop() {

}
```

#### http.cpp

```cpp

#include "http.h"

void IotHTTPClass::onSTAGotIP(WiFiEventStationModeGotIP ipInfo) {
  Serial.println("Got IP");
}

```

#### http.cpp

```cpp

#ifndef _IOT_HTTP_h
#define _IOT_HTTP_h

#if defined(ARDUINO) && ARDUINO >= 100
#include "Arduino.h"
#else
#include "WProgram.h"
#endif

#include <ESP8266WiFi.h>

class IotHttpClass : public IotClass {
public:
	void onSTAGotIP(WiFiEventStationModeGotIP ipInfo);
};

extern IotHttpClass HTTP;

#endif
}

```

### Debug Messages

```
error: no matching function for call to 'ESP8266WiFiClass::onSTationModeGotIP(<unresolved overloaded function type>)'
```

igrr, 2017-01-23:
```c++
WiFi.onStationModeGotIP(HTTP.onSTAGotIP);
``` 
is not valid C++. You need to use something like
```c++
e1 = WiFi.onStationModeGotIP(std::bind(&IotHttpClass::onSTAGotIP, HTTP));
```

czopas, 2017-01-23:
```cpp
e1 = WiFi.onStationModeGotIP(std::bind(&IotHttpClass::onSTAGotIP, HTTP));
```
Still not working. Message from console:

```cpp
error: no matching function for call to 'ESP8266WiFiClass::onStationModeGotIP(std::_Bind_helper<false, void (IotHttpClass::*)(WiFiEventStationModeGotIP), IotHttpClass&>::type)'
```

I am not very strong in C++ and I have no idea how to resolve this problem in my code.

hreintke, 2017-01-23:
@czopas 
You are missing the placeholder : 
e1 = WiFi.onStationModeGotIP(std::bind(&IotHttpClass::onSTAGotIP, HTTP,std::placeholders::_1));

czopas, 2017-01-23:
@hreintke 

With placeholder it works :)

Could you explain how it works and give an advice which version (static function or std::bin with placeholder) is more professional ?

hreintke, 2017-01-23:
@czopas : 
There is a lot of reading on the web for std::function, std::bind and lambda's.
On static/placeholder. I think the class/implementation decides whether a method is static or not.
Don't know/did not test, but it can be that the std::placeholders::_1 also works with static -> if so : one solution for both cases.

czopas, 2017-01-23:
@hreintke

Thanks a lot for help

closed 2017-01-23 by czopas