esp8266/Arduino #2895: analogWrite is not compliant with standard Arduino API

Makuna, 2017-01-23:
The standard Arduino API `analogWrite()` takes a value from 0-254.  The Esp Arduino version defaults to a 10 bit value.  This makes code incompatible between all other Hardware and leads to confusion on the part of users.

`analogWrite()` should be modified to default to 8bit and the standard `analogWriteResolution()` should be implemented to set the resolution to 10bit (and/or what ever else it can support).

Paul-Ver, 2017-01-24:
#295

Use the defined PWMRANGE.

You could then go along and calibrate it accordingly: https://www.arduino.cc/en/Tutorial/Calibration

devyte, 2017-09-09:
@igrr is this something that we want to do? I _really_ don't like defaulting to 8bits and having to change to 10bits with a function call, it would break who know what out there, but the request has a point about compatibility with Arduino.

igrr, 2017-09-09:
Maybe put this under "major breaking changes for new major release" category (i.e. if/when we bump the version to 3.0). I agree that this should be made Arduino compatible, at some point.

Makuna, 2017-09-09:
@devyte It currently is breaking stuff all over the place.  Several libraries don't work out of the box with esp8266 due to this (internally they use analog write).

igrr, 2017-09-10:
@Makuna can you mention some of these libraries? As in, it would probably take ~15min per library to send a PR to take PWMRANGE into account. That would be a reasonable short term solution, I think.

devyte, 2017-10-04:
@Makuna will you be providing the requested info?

Makuna, 2017-10-04:
The library issues came up in a maker class I was teaching where students applied specifics from their projects.  We discovered the issues with several, but I don't have the links to them sorry.

majenkotech, 2018-11-11:
It is confusing for an Arduino user that expects `analogWrite()` to be 8 bit. 

* [Here is one example](https://arduino.stackexchange.com/questions/57718/passive-buzzer-works-with-analogwrite-but-not-with-digitalwrite-it-also-ha) where they are confused that `analogWrite(pin, 255);` doesn't act the same as `digitalWrite(pin, HIGH);`.

PR #7456 (2020-07-29): BREAKING - analogWriteRange 8-bit default
Matching standard Arduino cores, make the default analogWrite() take
values from 0...255.  Users can always use the analogWriteRange() call
to change to a different setup.

Fixes #2895
commit a6798691 (2020-07-29): BREAKING - analogWriteRange 8-bit default (#7456)

Matching standard Arduino cores, make the default analogWrite() take
values from 0...255.  Users can always use the analogWriteRange() call
to change to a different setup.

Add a `analogWriteResolution` which takes a number of bits and sets
the range from 0...(1<<bits)-1, part of the standard Arduino API.

Remove the PWMRANGE define.  It's non-standard and not generally valid
(i.e. it's fixed at 1024 of 256, but the real range varies depending on
what you last set).

Also add note about the change and how to fix pre 3.0 applications.

Fixes #2 [... cut]
closed 2020-07-29 by earlephilhower