esp8266/Arduino #2897: WiFi access point disappears after a while

vahidyou2004, 2017-01-24:
I made a lot of boards using ESP07. It creates an access point and listens for connections on port 80. Some boards works well and some not. The WiFi access point that is created disappears after a while, even if there is a connection to it or not.
I used the below code to make the access point:
```
WiFi.mode(WIFI_AP);
WiFi.softAP(ssid, password);
```

Paul-Ver, 2017-01-24:
Can you open up the serial monitor and check if it prints anything?

Also put a println after setting up serial in the begin of your sketch (before attempting any network connections). Possibly power issues or something specific to your board.

If there seem to be power issues, you could use reservoir capacitor (reasonably big electrolytic cap) to rule out any power inconsistencies.

vahidyou2004, 2017-01-25:
I used a 12V5A power supply and a 1A LF33CV voltage regulator with a 100uF capacitor before it, and a 100uF capacitor and a 100nF MKT capacitor after regulator.

I turned serial debugging on and here is the output:
```
serial debuging started

loading default settings
loading settings from eeprom
initializing ports
initializing WiFi
bcn 0
del if1
usl
sul 0 0
mode : sta(5c:cf:7f:83:cc:af)
add if0
del if0
usl
sul 0 0
mode : softAP(5e:cf:7f:83:cc:af)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
creating access point
AP_83CCAF created
connect on 192.168.1.220
servers started on ports 80 and 3187
reseting login session
login session: 863761580333624132732820456
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
every 360 seconds check
reseting login session
login session: 86376153110633220132732820424
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
every 360 seconds check
reseting login session
login session: 86376151845861758132732820416
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
every 360 seconds check
reseting login session
login session: 8637615581090593132732820416
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
every 360 seconds check
reseting login session
login session: 86376153611288941132732820416
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
every 360 seconds check
reseting login session
login session: 86376152346515157132732820416
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
every 360 seconds check
reseting login session
login session: 86376151081744463132732820416
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
every 360 seconds check
reseting login session
login session: 86376154111943595132732820416
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
```
The AP disappeared somewhere about at the middle of above output. I just turned on the ESP and did not connected to it. Ass you can see the code is running after disappearing AP. The only problem is why AP disappears after a while on some boards?
Below output is a simple HTTP connection to ESP:
```
serial debuging started

loading default settings
loading settings from eeprom
initializing ihd
initializing WiFi
bcn 0
del if1
usl
sul 0 0
mode : sta(5c:cf:7f:83:cc:af)
add if0
del if0
usl
sul 0 0
mode : softAP(5e:cf:7f:83:cc:af)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
creating access point
AP_83CCAF created
connect on 192.168.1.220
servers started on ports 80 and 3187
reseting login session
login session: 863761580978997132732820456
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
add 1
aid 1
station: 00:18:de:8b:72:48 join, AID = 1
new HTTP client >> 
process post data ky=<not exists>
sending HTTP header
sending login HTML
sending footer
closing connection to HTTP client
new HTTP client >> GET / HTTP/1.1
process post data ky=<not exists>
sending HTTP header
sending login HTML
sending footer
closing connection to HTTP client
new HTTP client >> GET /favicon.ico HTTP/1.1
process post data ky=<not exists>
sending HTTP header
sending login HTML
sending footer
closing connection to HTTP client
every 360 seconds check
reseting login session
login session: 86376153111282835132732819168
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
station: 00:18:de:8b:72:48 leave, AID = 1
rm 1
every 360 seconds check
reseting login session
login session: 86376151846509269132732820224
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
```
Is this a problem with some ESP modules that maybe made not well?

vahidyou2004, 2017-02-01:
Any idea?

devyte, 2017-10-04:
@vahidyou2004 As far as I can tell, the AP is stable in the latest git after 24h running, on ESP-12F. I've tested on Nodemcu, Wemos D1 mini v2, and WittyCoud, 46 boards in total, and no issues found.
I therefore think you have (had?) a hardware issue. 
There is one other thing that comes to mind: a LF33CV with a 12V input voltage means that you are dissipating quite a bit, so I would expect the regulator to get warm. If it's close to the ESP, you could have a heat problem. Even if it's not the regulator, you could still have a heat problem. Check whether the ESP gets hot.
Closing due to issue not in the core.

closed 2017-10-04 by devyte

Manchicor, 2017-10-11:
I´ve been having this same issue with a ESP-01. Problem doesn´t seem to be associated with voltage or temperature, both checked. The AP dissapears, but the code is still running. The radio also is running since the debug shows the **SoftAPModeProbeRequestReceived** events and the stations that made those requests.  Also, not a problem of running out of memory since there´s about 28K free heap all the time. Sketch tested with IDE 164 and 182 with ESP8266 2.0.0 and 2.0.3.

kalwinskidawid, 2025-04-11:
@Manchicor  @vahidyou Did you find any solution on your problem? 
I have exactly the same problem with one NodeMCU v2 series.