esp8266/Arduino #2899: IR Remote Control interface on esp8266

unreality, 2017-01-25:
In the ESP8266 datasheet (https://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf) it mentions an IR Remote Control Interface (section 4.8, page 16) on the MTMS (IO14) and GPIO5 (IO5) pins. 

Is this an actual 38khz signal generated by hardware we can use, or is it just a software-implemented? Has anyone used this interface before?

lrmoreno007, 2017-01-25:
It's the documentation of ESP8266EX no ESP8266.

ESP8266EX is a development board with an ESP8266 and IR Remote integrated. http://tech.scargill.net/wp-content/uploads/2015/04/ESP8266-Development-Board.jpg

Close this issue.

Regards.

duncan-a, 2017-01-25:
It's certainly fully implemented on the dev board, but I think the original
post is still valid, particularly in terms of the 38kHz...

lrmoreno007, 2017-01-25:
Try this lib...
https://github.com/markszabo/IRremoteESP8266

unreality, 2017-01-29:
IRremoteESP8266, and every other IR lib for ESP ive seen, just uses software bit banging/delayMicrosecond to generate a 38khz signal. That's all well and good, but I was hoping the ESP had a way in hardware to generate the signal, either via PWM or using a Timer etc. 

So im wondering what functionality is referenced by the manual in that section.

tablatronix, 2017-03-14:
Is is references in RTOS technical , but not in SDK.

https://github.com/espressif/esp8266-rtos-sample-code/tree/master/02Peripheral/Infrared

>
13.1.1. Transmitting
Users can use the following methods to transmit carrier wave:
• BCK of I2S
• 38KHz carrier frequency generated by WS pin
• Carrier wave generated by any GPIO via sigma-delta function. However, the duty ratio
of carrier wave generated by sigma-delta is around 20%, thus MTMS pin (GPIO14) is
suggested, for this pin can generate standard square wave at a carrier frequency of
38KHz and a duty ratio of 50% exactly.
In the sample codes, data transmission queue is generated via the DSR TIMER interface of
system FRC2, while a state machine driving the transmission of infrared data is also
generated.
Considering that the timing precision of transmitting NEC infrared code should reach a level
of µs, when initiating IR TX, system_timer_reinit should be invoked to improve the timing
precision of FRC2. In user_config.h, enable the definition of USE_US_TIMER, then interface
function os_timer_arm_us can be invoked to implement precise timing at the level of µs

```
#define GEN_IR_CLK_FROM_IIS 1
#define IR_GPIO_OUT_MUX PERIPHS_IO_MUX_MTMS_U
#define IR_GPIO_OUT_NUM 14 
```

tablatronix, 2017-03-15:
Theres a sigma_delta_gen_38k , but this sdk has a sigma delta unused,  not sure what the deal is with that or wnat that even means.

tablatronix, 2017-03-15:
The other option in rtos (according to example), was to use iis clock or dma since sdk now has i2s registers, i wonder if we can use i2s_set_rate 38k for carrier..

hotplot, 2017-09-20:
For anybody else who finds this issue, you can generate a 38 kHz signal in hardware by configuring the MTMS pin as the I2S clock signal and setting the frequency appropriately.

I have a code example here: https://github.com/hotplot/BlastIR/blob/master/src/ir_send.cpp

devyte, 2017-10-04:
This type of question and discussion is best held at a community forum.
Closing due to off topic, see #3655 .

closed 2017-10-04 by devyte