esp8266/Arduino #2900: Compiling errors ArduinoJson- lib using ECLIPSE Arduino (SLOEBER) plugin

3edc, 2017-01-25:
Hi all,
I'll try to compile a string sample of this lib 'https://github.com/bblanchon/ArduinoJson' for an ESP, but get a lot of compile errors using the ECLIPSE Arduino plugin (SLOEBER). Within the ARDUINO IDE everything works fine.

So its perhaps not a ESP specific problem, but I have no idea where to request for assistance, the creator of` the lib or the creator of the ECLIPSE plugin.

Here is one of the the several compile errors:

fuzzer.cpp 

`#include` <ArduinoJson.h>
`class memstream : public std::istream {
`struct membuf : std::streambuf {
`membuf(const uint8_t *p, size_t l) 
`{setg((char` *)p, (char *)p, (char *)p + l);
`}
`};

Resulting in  these compilation errors:

...\fuzzer.cpp:3:39: error: expected class-name before '{' token
 class memstream : public std::istream {

...                                      ^
...ArduinoJson-master\fuzzing\fuzzer.cpp:4:34: error: expected class-name before '{' token
   struct membuf : std::streambuf {

Other libs are compilable, but this one causes trouble. 

Thanks for any help,
Gerry

hreintke, 2017-01-25:
@3edc : 
Do not know whether it is a ArduinoJson/Sloeber/ArduinoESP issue but the workaround is not to use the latest 5.8.2 version but the 5.8.0. That is the version which does not yet include fuzzer.

3edc, 2017-01-25:
@hreintke
Thank you very much for your help. Indeed it compiled and linked instantly. 

WTH is that 'fuzzer' that causes such problems ? 
br
Gerry

closed 2017-01-25 by 3edc