esp8266/Arduino #2901: #define STATUS - the same name of constantly was useed on ESP8266/ESP32 in ADS1256.h library

kamil2234, 2017-01-25:
### Basic Infos

This code below works  correctly on Arduino UNO/Leonardo but not on ESP8266/ESP32 platform.

#### Hardware
Hardware:			?ESP-12E/ESP32 dev boards/D1 R2 WiFi ESP8266  WeMos

### Settings in IDE

Module:  			ESP8266
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:	      qio
Flash Frequency:	40Mhz?
Upload Using:		OTA / SERIAL / USB
Arduino 1.8.1

### Sketch

// Arudino Sample Code to use ADS1256 library
// Read 4 sensor using effiecient input cycling
// Written by Adien Akhmad, August 2015

#include <ADS1256.h>
#include <SPI.h>

byte DRDYPIN = 2;
byte CSPIN = 7;

// set pins
//#define SELPIN 7 //
#define DATAOUT 12 // MISO
#define DATAIN 11 // MOSI
#define SPICLOCK 13 //

float clockMHZ = 7.68; // crystal frequency used on ADS1256
int lo = 300;
float myarray[300];

// Define ADS1256 PIN that is connected to Arduino
ADS1256 adc(clockMHZ,DRDYPIN,CSPIN,2.5); // RESETPIN is permanently tied to 3.3v

/* Variable to store sensor reading
sensor1 = connected on AIN0 - AIN1
sensor2 = connected on AIN2 - AIN3
sensor3 = connected on AIN4 - AIN5
sensor4 = connected on AIN6 - AIN7
*/

float sensor1, sensor2, sensor3, sensor4;

void setup()
{
Serial.begin(9600);
delay(10000);
Serial.println("Starting ADC");
delay(1000);
// rozdzielczość
//PGAmulti_1 0.0000005960464832810452 //PGA-1 ±5V /16777215
//PGAmulti_2 0.0000002980232416405226 //PGA-2 ±2.5V /16777215
//PGAmulti_4 0.0000001490116208202613 //PGA-4 ±1.25V /16777215
//PGAmulti_8 0.00000007450581041013064 //PGA-8 ±0.625V /16777215
//PGAmulti_16 0.00000003725290520506532 //PGA-16 ±312.5mV /16777215
//PGAmulti_32 0.00000001862645260253266 //PGA-32 ±156.25mV /16777215
//PGAmulti_64 0.000000009313226301266331 //PGA-64 ±78.125mV /16777215

// start the ADS1256 with data rate of 15 SPS
// other data rates:
// ADS1256_DRATE_30000SPS
// ADS1256_DRATE_15000SPS
// ADS1256_DRATE_7500SPS
// ADS1256_DRATE_3750SPS
// ADS1256_DRATE_2000SPS
// ADS1256_DRATE_1000SPS
// ADS1256_DRATE_500SPS
// ADS1256_DRATE_100SPS
// ADS1256_DRATE_60SPS
// ADS1256_DRATE_50SPS
// ADS1256_DRATE_30SPS
// ADS1256_DRATE_25SPS
// ADS1256_DRATE_15SPS
// ADS1256_DRATE_10SPS
// ADS1256_DRATE_5SPS
// ADS1256_DRATE_2_5SPS
//
// NOTE : Data Rate vary depending on crystal frequency. Data rates listed below assumes the crystal frequency is 7.68Mhz
// for other frequency consult the datasheet.

adc.begin(ADS1256_DRATE_30000SPS,ADS1256_GAIN_1,false);
Serial.println("ADC Started");

// Set MUX Register to AINO and AIN1
adc.switchChannel(0);
}

void loop()
{

for(int i=1; i<=lo; i++){
adc.waitDRDY(); // wait for DRDY to go low before changing multiplexer register
adc.switchChannel(0);
sensor1 = adc.readCurrentChannel(); // DOUT arriving here are from MUX AIN0 and AIN1
//Serial.println(sensor1,10);
myarray[i] = sensor1;
delayMicroseconds(150);
// delay(1);
}

for(int n=1; n<=lo; n++){
sensor1 = myarray[n];
Serial.println(sensor1,10) ;
// Serial.print("\t");
delay(50);
}

//print the result.
//Serial.print("data sensor 1: ");
//Serial.print(sensor1,10);

//Serial.println("");

delay(24000);
}

### Debug Messages

```

In file included from C:\Users\kamil\Documents\Arduino\libraries\ADS1256\ADS1256.cpp:7:0:

C:\Users\kamil\Documents\Arduino\libraries\ADS1256\ADS1256.h:11:16: error: expected unqualified-id before numeric constant

 #define STATUS 0x00

                ^

C:\Users\kamil\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32/rom/ets_sys.h:614:3: note: in expansion of macro 'STATUS'

 } STATUS;

   ^

exit status 1
Error compiling for board ESP32 Dev Module.

```

If I change name of constant in library i i don't have any problems with compilation this code and I can write it to flash.  

I  succesfully tried to upload this code to many other versions of ESP866 like (D1 R2 WiFi ESP8266  WeMos) , ESP8266 v01 and so on.   However my code still is not working on ESP8266 platform.  

Please let me know where is the issue.

devyte, 2017-10-04:
@kamil2234 this is caused by a name clash between the library and the Espressif SDK. We have no control over the SDK, because it belongs to Espressif, so there is nothing that can be done at this repo.
I suggest you open an issue against the author of the lib, requesting a change in that symbol, and overall compatibility with the ESP.
Closing due to unported 3rd party lib.

igrr, 2017-10-04:
Let's keep this open — I think we can safely remove that enum from the SDK headers, looks like it's not used anywhere.

devyte, 2017-10-04:
@igrr interesting, you are right. But if it's an sdk header, how do we avoid the symbol clash in future sdk releases? Do you intend to request the removal upstream? or do we keep a local patch to be applied on top of this file every time we get a new sdk release?
Or is this something like an override file that takes precedence?

igrr, 2017-10-04:
There are a few SDK headers which are being maintained in a modified state since the very first release of this project. Usually when updating the SDK I generate patches for SDK header files and then apply them to this project. Sometimes there are conflicts which need to be resolved manually, but usually it's not hard.

In this particular case I will also check with my colleagues if they can remove the enum from the SDK without breaking any existing applications.

devyte, 2018-02-28:
@igrr I have assigned to you, pending checking with your colleagues whether the enum can be removed.