esp8266/Arduino #2902: HTTPClinet https request fails if no fingerprint is provided

torntrousers, 2017-01-26:
As per the [comment](https://github.com/esp8266/Arduino/issues/2197#issuecomment-229364358) made after Issue 2197 was closed, it should be possible to use HTTPClient with an https url without a fingerprint.

olileger, 2017-01-29:
+1
Having the same request turns me to use WiFiClientSecure with more work to do

zolv, 2017-02-11:
Hi. As I checked, this issue should be fixed some time ago but is still there in 2.3.0-rc2.
@olileger Can You send some code snippet how to rewrite HTTPClient into WiFiClientSecure?

liebman, 2017-12-13:
This PR allows specifying '*' as the fingerprint to skip validation https://github.com/esp8266/Arduino/pull/3933

proppy, 2018-06-15:
Sorry for the late reply.

Maybe the proper solution is something like https://github.com/firebase/firebase-arduino/issues/341 ? What do you think?

proppy, 2018-06-15:
Oups commented on the wrong issue :)

devyte, 2018-07-12:
Currently bearssl is integrated. Although still experimental, it is strongly advised to migrate to it, because axtls is planned for deprecation, and then retirement. 
Closing this.

closed 2018-07-12 by devyte