esp8266/Arduino #2903: error in esp8266-07

AndroidDesigner, 2017-01-26:
Hi.I have a ESP8266-07. 
1. I soldered it to its breakout board.
2. I used the connection same as the below:
Gnd : gnd-uno
Vcc : 3.3 uno
Rx : Tx-uno
Tx : Rx-uno
GPIO0 : gnd
GPIO15 : gnd
3. I used from your core and Arduino IDE. but I will receive the following error in uploading:

### Debug Messages

```
Build options changed, rebuilding all

Sketch uses 222,197 bytes (23%) of program storage space. Maximum is 958,448 bytes.
Global variables use 31,572 bytes (38%) of dynamic memory, leaving 50,348 bytes for local variables. Maximum is 81,920 bytes.
C:\Users\pc\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9/esptool.exe -vv -cd ck -cb 115200 -cp COM4 -ca 0x00000 -cf C:\Users\pc\AppData\Local\Temp\arduino_build_29935/Blink.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM4
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

```

rizvanhaider, 2017-04-24:
I was facing the same issue... tried with all ports again and again.. one port did not gave this error...

devyte, 2017-10-03:
@AndroidDesigner The espcomm open failed problem is documented. If what the docs describe doesn't work, you either have bad wiring or a bad board.
Closing due to documented.

closed 2017-10-03 by devyte