esp8266/Arduino #2907: Connecting to ESP8266 web server

pauloaaf, 2017-01-29:
I made a web server using ESP8266 with the following IP 192.168.1.41, that IP address redirects to a web page with 4 buttons to controll 2 relays.
I am facing a problem where in my main computer i can controll the relays without any problem but in the other devices (smartphone, tablet and another computer) i can't connect to that IP, if i restart the module i can access using those devices but after 2 or 3 min i lose access and i need to restart the ESP8266 again.
While this happens, in my main computer i can always connect to the web server.
Why does this happen?
PS: Every device is connected to the same network.

nitrique, 2017-01-30:
Hi Pauloaff,

I have the same problem as you, I think : 

1 - Wifi etablished and server up and runing, no other port listen, just port 80.
2 - After various time (between 15 and 30 minutes), the ESP stop answer to iOS (On safari XMLHttpRequest).
3 - My program restart web server every 5 minutes (to try to keep connection) but no response.
4 - Response ok from firefox on windows 10
5 - Once the response is sent to windows device, the iOS can again access the ESP web page.
6 - loop to number 2

Like Pauloaff, when ESP has soft or hard reset, it answer again.

Thanks for help

kapyaar, 2017-01-31:
Have you thought of caching issue? Just putting the idea up there as a brain spark. Few years back, I was working on an embedded webserver. I have a webpage, and from that page, there is ajax calling for data to this server every seconds. After sometime, it will stop working. And then in my case, I can open another browser window or tab and it would work. I did not have to restart the device itself. Stackoverflow folks suggested about this cache possibility. So I tried ajax with a 
"?q=math.rand()" query so each query is different. This solved the issue for me. Not sure if it is same. Putting it out there so you can eliminate that possibility.

MrAda, 2017-01-31:
Can any of you guys help me even try to get my IP address via DHCP?  I  have tried all the examples I can find and they don't work.  I can see that my device is recognized by my router just that the my code is not sensing that it is connected.

nitrique, 2017-01-31:
@kapyaar sounds good for me, I'm trying to add header to firmware because pushing to Apple's validation is to long.

I added the following (if inApp safari take care): 
`
webServer.sendHeader("Cache-Control", "no-cache, no-store, must-revalidate");
webServer.sendHeader("Pragma", "no-cache");
webServer.sendHeader("Expires", "-1");
`

I reply if I find something working.

pauloaaf, 2017-01-31:
@kapyaar I did not understand that match.rand() part. Can you explain in a different way?
Also here is the code i have https://github.com/pauloaaf/ESP8266./blob/master/Leds.ino.

nitrique, 2017-01-31:
@pauloaaf It simply change the request to never be the same, cause if variable change, the cache is invalidated in client browser

AbrahamEsensors, 2017-01-31:
I just happened to be online :) math.rand() is actually a javascript function. If you are familiar with ajax, when you make a request to the server, you just add a random number so each query is different. This way browser wont cache.  I looked at your code. Equivalent method would mean your href will be something like
socket2On?q=random_number
socket2Off?q=random_number

Hope that gives the generic idea. But this will need a bit more coding as now you are hardcoding the html. You may need to use some js.

pauloaaf, 2017-01-31:
Thank you for the answers, i made this change on my code:
https://github.com/pauloaaf/ESP8266./blob/master/Leds_teste.ino
I think this is what you guys meant everything works fine in my computer but i can't even connect to the ip 192.168.1.41 through my smartphone. I don't think that is the problem.
Thank you once again for your help.

MrAda, 2017-02-01:
@pauloaaf Thanks for your example.  It helped me connect to my router.

pauloaaf, 2017-02-01:
@MrAda Glad you made it.

pauloaaf, 2017-02-01:
Know i tried a web page just saying Hello World, so there is no action here. And happens the same.

nitrique, 2017-02-02:
I find this interesting (if you use Apache Cordova) : http://www.plugreg.com/plugin/zazoomauro/cordova-disable-nsurl-cache

Or try disabling NSUrl cache.

Then, because ESP does only allow one connection at the time, try closing it as soon as possible when job done to permit other one to talk.

santhosh8438, 2017-07-12:
HI ,
I'm having slow connection issue when accessing my ESP8266-12E web server page (Capital Portal Advanced),
port IP is 192.168.4.1(default) using ARDUINO  (Capital Portal Advanced) (windows 7)

Issues are,
1. When i try to connect it to the Hotspot it's taking more time to connect than normal time.
2. After connection i'm accessing 192.168.4.1 page in browser but it's take very long time to appear and the SSID and PASSWORD credential input box takes longer time to appear.

can someone help!!!!

My CODE is Arduino default code
you can find Examples -> DNSServer -> Capital Portal Advanced (in Arduino)
Thank You!

devyte, 2017-10-03:
@pauloaaf please try PR #3362  with lwip2.

bill-orange, 2017-12-03:
AbrahamEsensors, I have found the discussion of adding ?q=match.rand().

In using Safari, I hang up on the call to temperatureGraph.js.  Where would I put the random number generator to test your solution in this code?

https://1drv.ms/u/s!AmXKqAwyCrbxh9xghTbwLTUu1Y8ZhA

AbrahamEsensors, 2017-12-03:
@bill-orange it is math.rand() to be precise. This happens only if you have continuous ajax calls from the browser to a server and after a few calls, the browser starts caching the results. I am not sure if this is your case. In the link you provided, I did not find any ajax calls. So it should load as such. Now, you said safari hangs? do you know where it hangs? is it the code? or code is fine but no data returned?

bill-orange, 2017-12-03:
The HTML works fine on Chrome and Firefox. On Safari it works on the first access after a cold start. On subsequent calls it hangs when “Loading” is displayed. This would be when it tries to load temperatureGraph.js. That’s about all I know, except that if I try it too many times fsbrowser becomes unresponsive.

devyte, 2018-07-13:
Original post sounds like the AP isolation issue, where the MAX is connected via wire to the router.
Even if that's not the case, several fixes have been merged for WiFiClient and lwip glue since this was last discussed, and there has been successful heavy testing.
Closing due to age. If there is still a problem, please open a new issue and fill out the required info.

closed 2018-07-13 by devyte

namita20, 2019-04-03:
CONNECT MY ESP8266 WITH MY MOBILE HOTSPOT AND TRYING TO GET RESPONSE ON WEBSERVER ON ANOTHER LAPTOP USING ANOTHER MOBILE HOTSPOT BUT ITS NOT WORKING

kapyaar, 2019-04-03:
It probably wont work. If you are working locally then the second hotspot has no idea of the first hotspot's connected devices unless you have somehow bridged them.