esp8266/Arduino #2908: section `.text' will not fit in region `iram1_0_seg'

olileger, 2017-01-29:
### Basic Infos

#### Hardware
Hardware:			ESP-8266
Core Version:      	2.3.0

### Description
I try to write my own firmware for my WeMos d1 mini.
I got a main *.ino file that calls a manager.
This manager handles all the application : setup, loop, calls to dependencies (DHT sensor, Wifi controller and so on).
I've written this manager in C++.
Today I tried to add few lines of code to add new features but I'm stopped by a linker error : `section `.text' will not fit in region `iram1_0_seg'`
I've `ICACHE_FLASH_ATTR`-ed all my function but unfortunately it has no solved my issue so I wonder to know what am I doing wrong.

### Settings in IDE
Module:  			Generic ESP8266 Module
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL (USB)
Compiling env :  			Windows 10, Arduino 1.6.13 (CLI mode)

### Sketch
My firmware is available at [https://github.com/olileger/iot/tree/master/esp/ESP8266_Firmware]

### Debug Messages
```
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
c:/users/XXXXXX/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2
/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\XXXXXX\AppData\Local\Temp\arduino_build_110175/ESP8266_Firmware.ino.elf section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

zgoda, 2017-01-30:
```
Compiling .pioenvs/d1_mini/src/main.o
Linking .pioenvs/d1_mini/firmware.elf
Calculating size .pioenvs/d1_mini/firmware.elf
Building .pioenvs/d1_mini/firmware.bin
text	   data	    bss	    dec	    hex	filename
319810	  10468	  30056	 360334	  57f8e	.pioenvs/d1_mini/firmware.elf
```

I guess you should make it a bit smaller to fit on ESP.

olileger, 2017-01-30:
thanks for your feedback zgoda.
Shoud I use C instead of C++ ? Maybe using the STL is a wrong way.
Btw before having the error the compilation process told me using 49% of memory.

zgoda, 2017-01-30:
I don't use too much C++ features in my code besides simple classes & compositions so I can't really recommend anything.

BTW 1: Arduino IDE is not that good in detecting memory usage on foreign chips & cores.

BTW 2: you don't need to call `ICACHE_FLASH_ATTR` when using Arduino core for ESP, this is done automatically.

olileger, 2017-01-31:
good to know many thanks @zgoda :)

everslick, 2017-02-17:
first of all, make sure you put all your constant strings into flash memory, by wrapping them in F() and PSTR() macros. example: String foo = F("foo"); this will free up space in the .text section.

igrr, 2017-02-17:
Moving strings into flash should not change utilisation of .text, because strings are placed into .rodata by default. .rodata and .text are placed into different regions of memory (DRAM and IRAM), so freeing up space in DRAM will not make more room in IRAM.

everslick, 2017-02-17:
ok, thanx for making this clear.

devyte, 2017-10-03:
@olileger is this still valid with 2.4.0-rc1 or latest git?
I don't think this is an issue with the repo. However, I looked at your code briefly, and despite certain bad practices, I don't really see anythin obvious, so I have no idea why your .text is so big. I have an app with almost 10000 lines of C++ code myself, including use of std::vector and std::map, and I don't have a problem.
I suggest testing latest git, and also do a clean install of the entire Arduino IDE as well as the ESP boards and all libs. In other words, delete everything, and install from scratch.
Please report back here if the above works for you.

igrr, 2017-10-07:
I think this should be fixed by https://github.com/esp8266/Arduino/pull/2804 (which is available in master and in 2.4.0-rc1).

devyte, 2017-12-29:
Closing due to no feedback in over 2 months, age, and likely already fixed in master.

closed 2017-12-29 by devyte