esp8266/Arduino #2909: Writing to EEPROM hangs during write loop

Agustis88, 2017-01-29:
### Description

I have this script that is trying to write to the EEPROM.  It writes but then hangs without any error messages.  It simply hangs with the script below. It does not write anything after the commit command. What and I doing wrong?  I am also unsure of the memory size.  I have tried 512, and 4096 and get the same behaviour.

### Settings in IDE

Module:  		Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### ino file

#include <EEPROM.h>

const int _EEPROMSTARTING = 0;
bool _debug=true;
const int _EEPROMLENGTH=512;

void DEBUG(String text) {
    if (_debug) {
        Serial.print("*Test: ");
        Serial.println(text);
    }
}

void DEBUG(String text, String param) {
    if (_debug) {
        Serial.print("*Test: ");
        Serial.print(text);
        Serial.print(": ");
        Serial.println(param);
    }
}

String WriteToEEProm(String value, int startingPoint)
{
    DEBUG("Writing to EEPROM", value);

    for (int i = startingPoint ; i < value.length() ; ++i)
    {
        EEPROM.write(i, char(value[i]));
        DEBUG("Writing: ", String(value[i]));
    }

    DEBUG("Commiting from EEPROM");
    EEPROM.commit();
    Serial.println("Completed");
    DEBUG("Finished writing to EEPROM");
}

void UpdateKey(String value)
{
    DEBUG("UpdateKey");
    DEBUG("Writing value to EEProm in UpdateKey: ", value);
    WriteToEEProm(value,_EEPROMSTARTING);
    DEBUG("After WriteToEEPROM");
}

void setup()
{
    Serial.begin(115200);
    EEPROM.begin(_EEPROMLENGTH);
    
    UpdateKey("123456789012345");
    DEBUG("After UpdateKey");
}

void loop()
{
}

### OutPut

�*Test: UpdateKey
*Test: Writing value to EEProm in UpdateKey: : 123456789012345
*Test: Writing to EEPROM: 123456789012345

dandandandandandandan, 2017-01-30:
I'd try this, also removed the EEPROM.begin from the setup.

`String WriteToEEProm(String value, int startingPoint)
{
  DEBUG("Writing to EEPROM", value);
  EEPROM.begin(512);
  for (int i = startingPoint ; i < value.length() ; i++)
  {
    EEPROM.write(i, char(value[i]));
    DEBUG("Writing: ", String(value[i]));
  }

  DEBUG("Commiting from EEPROM");
  EEPROM.commit();
  EEPROM.end();
  Serial.println("Completed");
  DEBUG("Finished writing to EEPROM");
}`

Agustis88, 2017-02-01:
yes that was it!  By moving the begin and commit and end to the same method helped.  I also removed the print lines as that was also causing the issue.  I also saw the other error I made.  Setting the method to a return of Type String and not returning.  Thanks so much

devyte, 2017-10-03:
Per last comment, closing as resolved.

closed 2017-10-03 by devyte