esp8266/Arduino #2910: WifiEvent Handling

tkerby, 2017-01-30:
### Description

Is there a way to use the WiFi Event Handling to recover the RSSI and Mac Address of received probe requests?

Specifically I would like to setup an AP using SoftAP and monitor for any probes it receives, logging the time, MAC and RSSI.

I've taken a good look at the WifiEvent example for a client but it doesn't cover how to access the data in the structures associated with the events. In particular I'm looking to read from this data in ESP8266WiFiType.h

+struct WiFiEventSoftAPModeProbeRequestReceived
+{
+    int rssi;
+    uint8 mac[6];
+};

Ideallu I'd like to expand this to both run as a dummy AP, but connect to an access point as a client to get the time from NTP and forward Probe requests to an MQTT server

tkerby, 2017-02-01:
@igrr I think you may be working on this. Has there been any progress?

igrr, 2017-02-01:
No, I'm not working on this issue. I don't know of a way to monitor probe requests in the non-OS SDK in SoftAP mode. So no clue how this can be implemented.

tkerby, 2017-02-01:
@igrr Apologies - I saw your comment here: https://github.com/esp8266/Arduino/issues/2100#issuecomment-224143871 and thought you were referring to passing the structures associated with events into the event handler so this data could be read.

I know it's possible in LUA and by using the SDK directly but I'd like to find a way to do this in Arduino

igrr, 2017-02-02:
Sorry — I haven't understood your question correctly last time. Opened #2917 which implements the method you need. If you're using the git version of ESP8266 Arduino core, you can give it a go by doing `git fetch origin && checkout feature/wifi_probereq_event`.

tkerby, 2017-02-04:
That works perfectly! Thanks very much for your help with this @igrr

igrr, 2017-10-06:
Thanks @devyte, indeed this has been implemented and is available in master. Applied the right labels.

d-a-v, 2017-12-13:
@tkerby's issue seems to be solved.
@devyte any feedback on your side ?

closed 2018-01-02 by igrr