esp8266/Arduino #2911: Is this community dead?

cimba007, 2017-01-30:
Just a stupid question .. but I noticed that over the last year the number of commits has drastically decreased. At the same time there are 76 pull requests?

Has anybody an explanation or some obvious point I might be missing?

igrr, 2017-01-31:
Yes, this is not a very useful question, and the answer is subject to interpretation.
Better question would be: hey folks, I see there's not much progress lately, is there something I can do to help? To which I could, for example, answer, that we need to finish integration of SDK 2.0 and set up automated builds of boards manager package to support 'nightly' packages :)

isnadh, 2017-01-31:
Polite and frank response (-.. . @igrr Just that some of us have turned esp8266 github page to our facebook. we check it everyhour of the day. Great work! Wish i am soo savvy i can contribute to this almighty little thingy.

closed 2017-10-03 by devyte