esp8266/Arduino #2913: ADC_MODE(ADC_VCC); crash compiler

pandacriss, 2017-01-31:
NTPClient.ino example (https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/NTPClient/NTPClient.ino) wont compile if i add ADC_MODE(ADC_VCC); at the beginning of the sketch. others examples compile normally

it print the following error messages:

_**NTPClient:132: error: 'IPAddress' was not declared in this scope
 unsigned long sendNTPpacket(IPAddress& address)
                             ^
NTPClient:132: error: 'address' was not declared in this scope
 unsigned long sendNTPpacket(IPAddress& address)
                                        ^
/tmp/arduino_modified_sketch_868703/NTPClient.ino: In function 'void loop()':
NTPClient:76: error: 'sendNTPpacket' cannot be used as a function
   sendNTPpacket(timeServerIP); // send an NTP packet to a time server
                             ^
/tmp/arduino_modified_sketch_868703/NTPClient.ino: In function 'long unsigned int sendNTPpacket(IPAddress&)':
NTPClient:132: error: 'long unsigned int sendNTPpacket(IPAddress&)' redeclared as different kind of symbol
 unsigned long sendNTPpacket(IPAddress& address)
                                               ^
NTPClient:132: error: previous declaration of 'long unsigned int sendNTPpacket'
 unsigned long sendNTPpacket(IPAddress& address)
               ^
exit status 1
'IPAddress' was not declared in this scope**_

if i comment/delete the ADC_MODE(ADC_VCC); it compiles again

#### Hardware
nodeMCU  ESP-12 80mhz
arduino ide 1.6.13
esp8266 libs 2.3.0
linuxmint 64 bits

pandacriss, 2017-02-01:
BTW why is sendNTPpacket() declared as unsigned long if it wont return anything?
line 131:  unsigned long sendNTPpacket(IPAddress& address)

pandacriss, 2017-02-04:
i found a workaround, thanks to tamptus in this thread https://github.com/esp8266/Arduino/issues/1759
> I had the same Problem with : ADC_MODE(ADC_VCC); and Version 1.6.8
My workaround:
Add a new Tab with file adc.h
with one line:
ADC_MODE(ADC_VCC)
add
#include "adc.h"
as first include in first tab.

devyte, 2017-09-09:
@pandacriss this is not an issue with latest git. Just make sure to place the macro at global scope outside of any function.
Closing.

closed 2017-09-09 by devyte