esp8266/Arduino #2914: ESP8266 Gets hot on boot and Wificonnect then resets

pauleilio, 2017-02-01:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

I have 3 boards all the same running my own sketch but one of them will struggle on a Wificonnect, get very hot then reset and I do not know why. The board does restart but the process happens again. When I say hot, too hot to touch. Has anybody else had overheating reset problem.

liquidfalcon, 2017-02-02:
If it's getting that hot, the board/module is almost certainly faulty and should be replaced.

gicho, 2017-02-02:
Another option is to check wiring - maybe short connection between pins,
incorrect power supply.
What board type is it - could you check or take a picture of the pcb -
there might be "production problems" like solder bridges, bad tracks, etc.

devyte, 2017-09-09:
@pauleilio this looks like a hardware issue, specifically faulty hardware.
Closing as off topic. Please refer to esp8266.com or stackoverflow if you need further discussion.

closed 2017-09-09 by devyte