esp8266/Arduino #2916: Race in interrupts??

andbjo, 2017-02-02:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	?2.1.0-rc2?

### Description

Problem description
In file: esp8266/cores/esp8266/core_esp8266_wiring_digital.c
In function: __attachInterrupt
the interrupt handler is set before cleaning interrupts for the receive pin witch make the receive sensitive to transitions high to low or low to high. 
When this happends the device gets stuck in boot phase. 

My proposal is to move 
the line `interrupt_handler_t *handler = &interrupt_handlers[pin];`  to after cleaning interrupt.
As:

extern void ICACHE_RAM_ATTR __attachInterrupt(uint8_t pin, voidFuncPtr userFunc, int mode) {
  if(pin < 16) {
//    interrupt_handler_t *handler = &interrupt_handlers[pin];
    handler->mode = mode;
    handler->fn = userFunc;
    interrupt_reg |= (1 << pin);
    GPC(pin) &= ~(0xF << GPCI);//INT mode disabled
    GPIEC = (1 << pin); //Clear Interrupt for this pin
// Put tis line here:
interrupt_handler_t *handler = &interrupt_handlers[pin];
    GPC(pin) |= ((mode & 0xF) << GPCI);//INT mode "mode"
  }
}

### Settings in IDE

Module:  			NodeMCU
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz
Upload Using:		SERIAL/USB
Reset Method:		nodemcu

/Anders

devyte, 2017-09-09:
@igrr if this is valid, I think we should give it priority, as it can cause a soft-brick.

igrr, 2017-09-12:
Moving the declaration of `handler` would not work like that, but thank you for pointing out the issue @andbjo. #3598 has been created to fix this. Please take a look (and try, if you can).

andbjo, 2017-09-12:
I know it was not the right way do do it, but it was a quick and dirty fix for me.

d-a-v, 2017-12-13:
@andbjo this issue seems to be solved by #3598, can it be closed ?

andbjo, 2017-12-13:
onsdag 13 december 2017 11.17.30 skrev  david gauchard:
> @andbjo this issue seems to be solved by #3598, can it be closed ?

I have not had time to test the new code yet. 
When I look at the code, it looks good.
So i think it can be closed.

-- 
My memory is like the Bermuda Triangle. Information sent there, can completely 
disappear without a trace.

/Anders

closed 2018-01-02 by igrr