esp8266/Arduino #2919: heap runs out of memory

Pfannex, 2017-02-02:
### Basic Infos

#### Hardware
Hardware:			ESP-12e
Core Version:      	2.3.0-rc2

### Description

using big arrays forced reboot, while heap info suggests that enough memory is available

===============================
call TEST      = 46760
run TEST       = 46736
Samples        = 290
ArraySize      = 18560
destruct TEST  = 43560

### Settings in IDE

Module:  			WeMos D1 R2 & mini 
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

//===> Librarys <===============================================================
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <AktSen.h>

//===> GLOBAL <=================================================================

//===> constants <--------------------------------------------------------------
const char* ssid = "Pf@nne-NET";
const char* password = "Pf@nneNETwlan_ACCESS";
const char* mqtt_server = "192.168.1.203";

//===> variables <--------------------------------------------------------------
WiFiClient espClient;
PubSubClient client(espClient);
IPAddress ip;

//===> SETUP <==================================================================

void setup() {
  //RS232
  Serial.begin(115200);
  
  //ESP8266
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
}

//===> WIFI SETUP <-------------------------------------------------------------
void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  ip = WiFi.localIP();
}

//===> SUB-Routines <===========================================================

//===> MQTT callback <------------------------------------------------------
void callback(char* topic, byte* payload, unsigned int length) {
  char value[500] = "";
  for (int i = 0; i < length; i++) {
    value[i] = payload[i];
  }
  
  //Serial.println(topic);
  
  if (strcmp(topic, "ESP8266_1032096/ADE7953/getValues/getWaves") == 0){    
    Serial.println("===============================");
    Serial.println("call TEST      = " + String(ESP.getFreeHeap()));
    //TEST(String(value).toInt());
    TEST(String(value).toInt());
    Serial.println("destruct TEST  = " + String(ESP.getFreeHeap()));
  }

}
//===> MQTT publishTEST <------------------------------------------------------
void TEST(int samples) {
  Serial.println("run TEST       = " + String(ESP.getFreeHeap()));
  Serial.println("Samples        = " + String(samples));
  Serial.println("ArraySize      = " + String(2 * 32 * samples));

  unsigned long t[samples];
  uint32_t values[samples];
  unsigned long t0 = micros(); 
  for (int i=0; i<samples; i++){
    t[i] = micros();
    values[i] = random(100);
  }
  
  //String wave = "";
  for (int i=0; i<samples; i++){
    char cTime[30] {""};
    char cValue[15] {""};
    dtostrf(0.001*(t[i]- t0) , 1, 5, cTime);
    dtostrf(values[i], 1, 5, cValue);
    
    strcat(cTime, ",");
    strcat(cTime, cValue);
    
    client.publish("ESP8266_1032096/ADE7953/values/V_INST", cTime);
    client.loop();
  }  
}

//===> MQTT reconnect if lost <--------------------------------------------------
void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP8266Client")) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      //client.publish("outTopic", "hello world");
      // ... and resubscribe
      client.subscribe("ESP8266_1032096/#");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

//===> MAIN-LOOP <===========================================================

void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
}
```

### Debug Messages

```
===============================<\r><\n>
call TEST      = 46760<\r><\n>
run TEST       = 46736<\r><\n>
Samples        = 290<\r><\n>
ArraySize      = 18560<\r><\n>
destruct TEST  = 43560<\r><\n>
<\r><\n>
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)<\r><\n>
<\r><\n>
wdt reset<\r><\n>
load 0x4010f000, len 1384, room 16 <\r><\n>
tail 8<\r><\n>
chksum 0x2d<\r><\n>
csum 0x2d<\r><\n>
v3de0c112
```

liquidfalcon, 2017-02-03:
You're not allocating your arrays on the heap, they're on the stack. The Arduino ESP8266 project has a default stack size of 4K, and you're almost certainly overflowing that. Either put it on the heap, increase the stack size, or reduce the array size.

Pfannex, 2017-02-03:
OK!
Global uses heap and local uses stack (4k)?
how to increase the stack?

putting the array to global works!

liquidfalcon, 2017-02-03:
Generally that is the rule, yes. You can manually set the stack size by modifying CONT_STACKSIZE in cores/esp8266/cont.h 

You can get a rough idea of free stack space by calling `cont_get_free_stack` in that same header file. I think you're better off malloc'ing and free'ing the memory, or keeping it global, than playing around with the stack size - Assuming you can't instead just refactor your function instead to not require such a huge chunk of memory.

Pfannex, 2017-02-03:
> why do you need to build all samples before you run your publishing loop? 

The first loop reads samples from a I2C-ADC an I need SampleRates up to 3kHz.
Doinf this in one loop the SampleRate is to Slow.....

```c
  int sampleRate = read(SampleRate);  //in kHz
  unsigned int tsample = 1000000.0 / (sampleRate * 1000); //in µs
  int samples = read(Periods) * 20000 / tsample;
  
   for (int i=0; i<samples; i++){
    values[i] = read(regNumber);
    delayMicroseconds(tsample-106);
  }
  
  String wave = "";
  for (int i=0; i<samples; i++){
    char cTime[30] {""};
    char cValue[15] {""};
    dtostrf(0.000001* tsample * i, 1, 5, cTime);
    double val = getFullScaleInput(read(PGA))* sqrt(2) * uint24Tolong32(values[i]) / 6500000.0 * read(k) / 100;
    dtostrf(val, 1, 5, cValue);
    
    strcat(cTime, ",");
    strcat(cTime, cValue);

    if (callback != nullptr) callback(cTime, regNumber);
  } 
```

devyte, 2017-09-09:
Closing this as user error (big stack usage).

closed 2017-09-09 by devyte