esp8266/Arduino #2921: How can i send post request to a jsp Page?

drakehammer, 2017-02-03:
Hi to everyone, im making a trilateration app using wifi as technology and im usign esp8622 as tag, i made a java code with apache commons math to make all the calculus in android, but now i want to migrate it to web, and i dont know if it possible to send the post from the esp8622 to a jsp page.

Thanks for your help.

wuwx, 2017-02-03:
http.POST(payload);

drakehammer, 2017-02-03:
I was trying, but i cant use it, can you help me with a example? please

torntrousers, 2017-02-04:
There's one [here](https://github.com/esp8266/Arduino/issues/1390#issuecomment-170343346).

devyte, 2017-09-09:
@drakehammer there are examples viewable from within the Arduino IDE.
This is not the right place to ask for assistance. This is an issue tracker, meant to track issues in the core. Please refer to esp8266.com or stackoverflow for assistance, assuming you still need it.
Closing.

closed 2017-09-09 by devyte