esp8266/Arduino #2922: Problem with WiFi not able to connect: STA disconnect: 201

vks007, 2017-02-03:
### Basic Infos
Connecting to WiFi has intermittent issue of  STA disconnect: 201
#### Hardware
Hardware:			ESP-01
Core Version:      	2.3.0

### Description
I am having trouble with the ESP 01 connecting to WiFi in certain scenarios. The problem does not happen always but when it does, I am not able to set it right. The issue is that it keeps on trying to connect to WiFi and keeps on disconnecting with the error messages as shown below. I am using this chip on a circuit I have wired on the PCB. I have the same circuit wired on the breadboard. The strange part is that I have not got this error even once on the bread board. It always works there. At times the problem went away when I re-flashed my code on the ESP and then again it comes back after some time. At times I have waited for a long time and it successfully connects after trying for 5-10 minutes. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <FS.h>

//#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

#define MAX_SWITCHES  4

//This program can control 4 switches in total, each switch has a input touch pin and an output pin
//Switches are numbered as 0,1,2,3 and thier ports are defined as in the array arrPinConfig
//eg. Switch 0 will have GPIO0 as touch pin and GPIO2 as output pin
// Switch 1 will have GPIO4 as touch pin and GPIO5 as output pin and so on...
//[touch pin no][out pin no]
const int arrPinConfig[MAX_SWITCHES][2] = {{0,2},{4,5},{12,13},{14,16}};
//Note that you cannot assign GPIO2 as input pin because during start up this pin will be held LOW by the touch switch. GPIO2 should be held high or flaoting during boot else the chip will not boot. 
//Read blogs on this for more info

bool switchStatus[4];//Stores the status of the output PIN, true means HIGH , false means LOW

//For a variable to be accessible in ISR routine and main program it has to be declared as volatile
volatile bool changeState[MAX_SWITCHES] = {false,false,false,false};
volatile short switchNo[MAX_SWITCHES];

const char *ssid = "EAGLE"; //SSID of WiFi to connect to
const char *password = "password"; //Password of WiFi to connect to

ESP8266WebServer server ( 80 ); //creates a web server at port 80

/******************************************************** SETUP ********************************/
void setup ( void ) {
  Serial.begin ( 115200 );
  Serial.println("");
  //Set the pin modes for the touch pin and the putput pin
  for(int i=0;i<MAX_SWITCHES;i++)
  {
    pinMode(arrPinConfig[i][0],INPUT);//These are all touch pins
    pinMode(arrPinConfig[i][1],OUTPUT); //These are all output pins
  }

  //Read the last stored value from the EEPROM and set the output pin accordingly
  //Note that this will behave randomly when the program runs for the first time, I dont intend to handle that
  // initialize the SPIFFS
  if (!SPIFFS.begin()) 
     while(1);

[... 387 lines cut ...]
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.....wifi evt: 1
STA disconnect: 201
......wifi evt: 1
STA disconnect: 4
......wifi evt: 1
STA disconnect: 2
.....wifi evt: 1
STA disconnect: 201
..........wifi evt: 1
STA disconnect: 15
....wifi evt: 1
STA disconnect: 4
...wifi evt: 1
STA disconnect: 2
wifi evt: 1
STA disconnect: 201
......wifi evt: 1
STA disconnect: 2
......
```
messages here
```

CircuitSerialKiller, 2017-02-06:
Where'd you get the ESP chip from? Can you try an ES module from a different source? Have you tried a different AP?

vks007, 2017-02-06:
I got ESP from a seller in China. I have tried multiple ESP chips and they all behaved this way many times. To further debug this issue I de-soldered all connections in the circuit except the ones needed by ESP and it worked fine. Then I started soldering one connection at a time. Although I cant say with certainty, the issue may be the way I am using the GPIO0 as an input. I have to supply the input from a touch module to this pin. I have the input from touch coming to the base of a transistor BC548 via a 1K resistor, emitter is grounded while I have taken the output form the collector. The collector anyway is connected to the 10K pull up at GPIO0. It was necessary to put the resistor because else at startup the touch module output is LOW pulling GPIO0 LOW and hence causing it to enter the boot loader mode. I was able to reproduce the issue most of the times with this and it always worked fine without the input from transistor. But then it was never deterministic. Sometimes it behaved fine even with the transistor. I again re-wired the stuff and now am able to reproduce it very few times. The fact remains that the same circuit on my breadboard has never experienced it.

devyte, 2017-09-09:
@vks007 you have a delay(5000) inside the loop. That means that if data comes in, the wifi stack can't pass it to the application layer, and could blow up. You should not have such big delays.
In addition, latest git moved to SDK 2.0.0, which I believe does RF calibration on boot, so connectivity is much more stable.
I am not aware of current connectivity issues. 
This looks like user error to me. Closing.

closed 2017-09-09 by devyte

vks007, 2017-09-09:
Hi @devyte , the program I pasted below is from the examples. I don't have a delay in my program, it still behaves the same way.  In fact any program I run behaves the same way.
Thanks for the inputs though.

devyte, 2017-09-09:
@vks007 BTW: 

>  I have not got this error even once on the bread board

should be your biggest hint. I think you have a hardware issue, your PCB is somehow interferering with the wifi. 
Still, try latest git, maybe the recalibration call improves things for you.

igrr, 2017-09-12:
@devyte regarding

> you have a delay(5000) inside the loop. That means that if data comes in, the wifi stack can't pass it to the application layer, and could blow up.

There shouldn't be a problem with delay in this case. It is called before TCP connection is established, so no TCP data can come in at that point. All other traffic will be handled by WiFi stack and LwIP, while the sketch is suspended until the end of `delay`.

(but otherwise i agree, looks like a board issue)

psatheesh18, 2018-02-22:
i have the same issue but code works well in nodemcu, but it is not working on the esp01 module

zhouyangwen, 2019-01-01:
the issue would be soulved when i compile it with board manager esp8266 community version 1.6.5-947

Ma7amed, 2019-09-07:
It is really weird, I just had this issue yesterday with esp8266-01. The circuit work normally in breadboard and prototype pcb, but fails to connect to wifi most of times in the production pcb.but I'm using gpio0 as input pin.

TheShubhendra, 2022-03-13:
I am having the same issue with my ESP01. It is able to connect with the Mobile hotspot but not with my airtel wifi router. I have tried the WifiScan example and it is able to scan the wifi but not connect.

d-a-v, 2022-03-13:
@TheShubhendra
it may have been solved with the latest dhcp fixes.
Try with the latest git version or the [unofficial snapshot release](https://d-a-v.github.io/esp8266/Arduino/index.html)

TheShubhendra, 2022-03-13:
> @TheShubhendra it may have been solved with the latest dhcp fixes. Try with the latest git version or the [unofficial snapshot release](https://d-a-v.github.io/esp8266/Arduino/index.html)

Thanks for your response. I have tried but the issue is still there.

Kishlay-notabot, 2024-02-20:
Facing the same issue, but its `sta disconnect 13` and `sta disconnect 202`

Kishlay-notabot, 2024-02-21:
Update: I changed the WPA settings from `WPA-2 personal' and it works.
It cannot connect to `WPA3` or `WPA2/3`...
Im able to configure my phone hotspot that way and it connects succesfully. 
But I cannot connect it to my router, which has `WPA2/3`