esp8266/Arduino #2923: Missing documentation of all persistently stored SDK config

objarni, 2017-02-03:
### Basic Infos
I need a list of all APIs storing things on the ESP8266 persistent memory to be able to reproducibly flash my production chips.

#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0

### Description

I have come to learn that the ESP8266 Arduino SDK stores at least WiFi.begin state, and WiFi.mode, on the chip 'somewhere'.

This means that if I exchange some code to some other code, e.g. with my friend, I cannot be sure exactly what state the chip is in when booting.

The "solution" is to *always* write all known persistently storing state in the SDK on boot, so that I have a reproducible behaviour.

However, as a embedded systems product developer, this really annoys me - an SDK should *not* store any state, that is up to the application developer. I can see the point for a "home toy project" since it makes it just slightly simpler for the programmer, but for any serious product development it makes it a lot harder.

At least, could we have an authoritative list of what state is stored, so that at boot all that state can be updated to a known config...? I have googled quite a lot on this topic, but found no such documentation page.

### Settings in IDE

Does not apply

### Sketch

D.N.A.

### Debug Messages

D.N.A.

objarni, 2017-02-04:
In 3D computer graphics, OpenGL is a popular cross-platform library. Storing state on persistent memory would, in comparison, mean e.g. that the OpenGL lib would store whether the depth buffer is on or off on the hard drive. I.e. a library user could not assume anything about the default state of the library/graphics card was when booting the application, but instead would have to set everything to a desired state.

Or an even simpler example: how float numbers are formatted using "cout" in C++ (a runtime state with a default state).

igrr, 2017-02-04:
Your concern about WiFi stack saving its state behind the scenes is a valid one, but it's outside of scope of this project. WiFi stack comes from the non-OS SDK for the ESP8266. It would make more sense to raise this issue against the SDK.

The point about documenting the behavior is also a valid one, and hopefully will be addressed when we start having reasonable documentation for libraries.

objarni, 2017-02-04:
OK, fair enough. Where is the the right place for this valid issue about missing documentation then, I'll file it there.

igrr, 2017-02-04:
Regarding documentation for Arduino, that place is here (this issue is ok). Regarding documentation for the SDK, you can submit it to bbs.espressif.com.

objarni, 2017-02-04:
(.. or tell me where the source code is and I'll try and figure out the state myself. It is open source I hope..? Should be enough to look at how e.g. WiFi.mode stores the enum value and go from there)

objarni, 2017-02-04:
"Regarding documentation for Arduino, that place is here (this issue is ok). " I'm confused -- is this the right place for this issue or not?

To be clear, I see this project, esp8266/Arduino as a C/C++ implementation of the Arduino API (SDK) for ESP8266 based SoC's, as opposed to the Arduino SDK which is the same thing but for Arduino SoC's. That is the view that made me assume the 'storing' is happening here and nowhere else.

igrr, 2017-02-04:
The stack (part of the ESP8266 SDK) isn't open source. This project is a C++ wrapper over the functions of the SDK. SDK functions store the state, the code in this project does not do anything other than calling SDK functions.

You have raised the issue about documenting Arduino API functions which store the state. It can be addressed within the scope of this project.

You have also raised an issue that saving of state should not happen. This can be achieved by calling `WiFi.persistent (false);`. However we don't have a way to avoid *loading* of state from flash — this happens somewhere in the depths of the WiFi stack. This issue has to be addressed in the ESP8266 SDK.

objarni, 2017-02-04:
@igrr Thank you for a thorough explanation.

If it is *only* WiFi credential state that is stored, the problem seems manageable. ATM I got a NodeMCU0.9 with unclear behaviour, and I managed to run some out-of-bounds code on it (access outside of c-array), which means it might very well have called completely random places in memory. I want to completely reset the flash memory to remove that possibility from debugging efforts.

devyte, 2017-09-09:
@objarni the config bytes stored in flash are documented in the espressif SDK api reference (document 2c if memory serves). Re-documenting it here doesn't make sense.
In addition, there are config bytes that you do NOT want to keep. Example: the RF calibration data, whic adapts to the surrounding environment every boot.
In any case, it is possible to read back the flash with the esptool for reproducability.
Considering that the concern seems to have been over the Wifi credentials, which can be set during setup, that the requested documentation already exists within the SDK docs, and that there has been no feedback in several months, I am closing this.

closed 2017-09-09 by devyte

objarni, 2017-09-09:
@devyte ok but this library implements the Arduino SDK, on top of espressif SDK, right?

IMHO any behaviour differences from vanilla Arduino SDK should be documented here, is the point of this wrapper library is to mimic Arduino SDK.

philbowles, 2017-10-31:
Is there an elephant in the room here? If your "transportable" code resets the "state" to a known starting point (i.e. zero / null / whatever, depending on what "state" means to your code...) then the problem is gone, non? Think of the way that many browser tool have a "reset" css. At its most abstract, what you are saying you need is a known (and common) starting point...by experimentation / testing you can find/ create / force one that works for your code.

The argument "the SDK _shouldn't_ do X...." is good for the classroom - the fact that it _does_ is a real-world problem that requires a workaround. Machiavelli said "We must live in the world the way it _is_, rather than the way it _should_ be".

I had a similar problem - some "edge cases" where serious tinkerers had been using some pretty rare WiFi commands would leave the device in a state that would cause my general-purpose library code code not to connect, where in 99% of "normal" cases is always did. Solution? I re-wrote my code to first reset everything that could prevent it working. It now works no matter what any previous state of the device may have been. No SDK changes required.

Do I care about the philosphical ugliness of an SDK that does things that make no difference to my code in the real world? No, I work round it. "_Anybody_ can fix for $1 what an engineer can fix for 1c"